imbushuo/mac-precision-touchpad #284: Uninstalling driver

mialkin, 2020-04-02:
I installed driver on my Windows 10 by right clicking AmtPtpDevice.inf file and selecting Install.

I am running Windows 10 on my MacBook Pro 2019 16'' as the second operating system under Boot Camp.

How can I remove the driver to fall back on default driver implementation? I don't like absence of palm rejection on this driver. Is it even possible now to fall back on default implementation of touchpad driver or I'll need to reinstall Windows?

Thanks in advance.

P.S. It would be nice if you included uninstalling instruction on main page next to installation instruction.

summatix, 2020-04-07:
I was able to uninstall it by going into Device Manager, under Human Interface Devices, right-click on Apple Precision Touchpad and then select uninstall. Make sure not to select disable first because that will keep the driver but disable the trackpad. By uninstalling you will fall back to the Apple driver.

imbushuo, 2020-04-18:
I've added uninstallation instruction at https://github.com/imbushuo/mac-precision-touchpad/blob/master/README.md.

closed 2020-04-18 by imbushuo