imbushuo/mac-precision-touchpad #288: Single click with one finger + move with another

citizenmatt, 2020-04-09:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
Using one finger to select and move something, like dragging a window, is hard and error prone. Clicking or tapping a window title bar and moving will quickly hit the side of the touch pad, requiring releasing and clicking/tapping again to move. Clicking works most often, but is not as comfortable as tapping (I much prefer tap to click). Tapping is nicer to use, but doesn't always register, causing frustration that the drag/drop is no longer working and the cursor is now in the wrong place.
An alternative that works in macOS is to use two fingers - one to tap/click and not move, and the other to move around the screen, but this causes the cursor to bounce around the screen, which is again very frustrating.

**Describe the solution you'd like**
Use two fingers to click/tap and move for drag/drop, window moving, etc. Click or tap with one finger, keep it still, then use another finger to move the cursor accurately. Picking this finger up from the trackpad should not send the cursor to some weird position

rbreaves, 2020-12-23:
I think I finally know what is going on here, I was about to open an entirely new ticket for a similar issue.. where highlighting text sometimes leads the selection jumping upwards wildly.. I think the issue is that it detects a held left click and then a double tap while the user is just trying to drag for a selection of text.

I think double tap behaviors or detection needs to go away when a left click is being held. Is this possible to do?

rbreaves, 2020-12-23:
Reading up on this more I may just have to retrain my brain to select differently so it doesn't go insane. Double tap and then drag to select, whether I keep holding pressure or not because perhaps Windows just isn't designed to select text in the same manner as macOS is able to do it.