imbushuo/mac-precision-touchpad #292: Trackpad not working in games

Salbawardi2, 2020-04-21:
I tried playing some games on my MBP 16 but the trackpad would freeze inside the game. Switching apps makes it work again, but I can't get it to work inside the game itself. 

I tried reinstalling the latest version but that didn't help. Regular apple drivers work and trackpad++.

Thanks!

DrJedikiah, 2020-04-24:
It's a bug of Microsoft I think because it also happens with my laptop which has native precision support.

Salbawardi2, 2020-04-26:
Any fix?

Bakaslam, 2020-05-13:
Following as well. 

Not sure if it's the same bug, but the trackpad would stop working until I press it down while playing.

davidfang00, 2020-05-13:
Same here. Stops working as soon as I open up a game.

Bakaslam, 2020-05-14:
I'd be happy even if just a temporary fix would be implemented. Something like disabling temporarily the software with a shortcut. The trackpad works great outside of games and if we could at least switch between the precision touchpad and classic touchpad  might be a good start.