imbushuo/mac-precision-touchpad #295: MacBook Air 2020 (MacBookAir9,1) Support

ShouriWiN, 2020-04-28:
Hello. I got to know this project by reading an article on a Japanese site. I wanted to try it, but apparently it doesn't work with the recently released MacBook Air 2020 (MacBookAir 9,1).

The installation was completed, but the display of Device Manager (Apple Multi-Touch) is the same, and the display such as "Precision ~" is not found.

In addition, I have attached images referring to past issues.
The trackpad size is probably the same as the 2019 model.

Bootcamp-Windows10, ver.1909, build.18363.815
MacBook Air built-in trackpad

Sorry for the long sentences!
Thank you.
_(Sorry. I'm Japanese and I'm not very good at English, so I use Google Translate to write it.)_

[image]

JeremyLee034963, 2020-05-05:
Same issue here. I really hope 2020 MacBook Air is included in the support list. Keep up the great work!

imbushuo, 2020-05-10:
@ShouriWiN こんにちは！ (Well I am learning Japanese now but just get started)

I've pushed an update to https://github.com/imbushuo/mac-precision-touchpad/commit/7aaa0448997013bcd41b731d4deac7f24e97fce8; please try package at https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3824&view=results and see if that works.

Note: you need to turn off Secure Boot as outlined in https://support.apple.com/en-us/HT208330 in order to get CI builds installed.

ShouriWiN, 2020-05-10:
@imbushuo こんにちは！！私はあなたが日本語を勉強していると知って、とても嬉しいです！！
そして、あなたの働きと返信に感謝します。

I tried it immediately, so I will report it.

(First of all, is there any mistake in downloading the file "Drivers-amd64-ReleaseSigned" among the five files in the "Related" column of the link destination? Sorry if you made a mistake.)

① I tried to install the "inf" file according to the Github procedure.
②A popup appears asking whether to allow the installation. (I did not come out until now)
③ I received the message that the installation was successful as before.
→ But nothing happened as when I posted this issue.

④ I tried to update the driver manually from Device Manager. It is written as Apple USB Precision Touchpad Device (Kernel-mode).
⑤ What about! It started moving! The touchpad item for setting is also displayed! 

[image]

I tried variously after that, but there is no particular problem and it seems that there are no gestures with 2 to 4 fingers and no dead zone on the touchpad.
(There was a problem that the option of right-clicking with two fingers did not appear until the touchpad was released.)

In summary, it did not work unless you manually selected the driver from Device Manager.
Moreover, I found that the build released on Github will work if the same steps are followed.

JeremyLee034963, 2020-05-10:
@imbushuo 

Hello,

I tried the new package, and it still doesn't work.

It is still named 'Apple Multi-Touch', not precision driver.

*I turned off Secure Boot.

I have no clue what is wrong with my MacBook Air.

imbushuo, 2020-05-10:
It’s expected that you have to manually select drivers from CI artifacts as @ShouriWiN pointed out. Anything from CI is signed by my certificate but not Microsoft, however, official releases on the GitHub release page are. So I am going to push a release this week to finalize the support.

imbushuo, 2020-06-19:
See https://github.com/imbushuo/mac-precision-touchpad/releases

closed 2020-06-19 by imbushuo