imbushuo/mac-precision-touchpad #298: How to download?

Bakaslam, 2020-05-11:
Sorry for the dumbest question ever, but how do I download this? 
I don't see any "release" tab as indicated in the main page, and if I click on "Clone or Download", I don't see AmtPtpDevice.inf file for install.

sm1ee, 2020-05-11:
You can download from [this](https://github.com/imbushuo/mac-precision-touchpad/releases) page.

Bakaslam, 2020-05-11:
> You can download from [this](https://github.com/imbushuo/mac-precision-touchpad/releases) page.

Thanks! Do I download all the 4 files, just one of them? And once I do, what do I do with them? Sorry for the second dumbest question :)

tech189, 2020-05-13:
Most likely just download the first one in the list, with amd-64 in the name, then unzip it and right click AmtPtpDevice.inf to install it. There should be an 'install' option in the right click menu.

Bakaslam, 2020-05-13:
> Most likely just download the first one in the list, with amd-64 in the name, then unzip it and right click AmtPtpDevice.inf to install it. There should be an 'install' option in the right click menu.

Thank you tech, followed your instruction and managed to install it. Is there anyway where I can access settings for this so to adjust sensibility, gestures, etc.?

tech189, 2020-05-13:
There was talk about at some point making a settings app, but for now I think just use the ones in the Settings app under Devices>Touchpad.

Bakaslam, 2020-05-13:
> There was talk about at some point making a settings app, but for now I think just use the ones in the Settings app under Devices>Touchpad.
> 
> Edit: I see you found that already, no worries!

Yep, thanks! 

As a suggestion, I'd say that for a newbie it's not clear if it's a piece of software kinda like Trackpad++ where you have a set amount of features that you can enable. From my understanding this software makes your trackpad be recognized by Windows as a "precision touchpad", which makes it possible to access the Windows native settings to customize it. Maybe a saddle difference, but an important one. 

I think it's brilliant, so thanks again!

closed 2020-05-24 by imbushuo