imbushuo/mac-precision-touchpad #299: bluetooth connectivity

miranemec, 2020-05-11:
I saw some request for BT in past, but there is no one in present. Is there any chance to get BT version? May I help you somehow? This driver is great, just BT is missing for me.

baltedewit, 2020-05-19:
Duplicate of https://github.com/imbushuo/mac-precision-touchpad/issues/229 ?

evanbebop, 2020-05-21:
I downloaded the latest version 3805 amd64 version. This problem also occurred on my mbp 15' late 2016. I uninstalled the original Apple trackpad from the device manager first and then installed the 3805 version. The trackpad on my mac works fine and magic pad also works fine through usb connection. However, it cannot be detected sometime through BT, and even if it is connected to windows, it just works like the original trackpad. The computer recognizes it as a mouse and all the function you have created do not work on this magic pad. Strangely, the trackpad on the mac works all fine(having all the function you create).

1Revenger1, 2020-05-23:
@evanbebop this driver doesn't have any bluetooth support yet

evanbebop, 2020-05-23:
Thank you very much. I just misunderstood the way to use this drive. Waiting for the further function. Will buy a coffee tomorrow.

Best wishes,

Zhuoran

获取 Outlook for iOS<https://aka.ms/o0ukef>

nrudnyk, 2020-06-01:
is there any plans to implement BT functionality?

karthi6662003, 2020-06-03:
I am using MacBook air 2017 .Built in trackpad works excellent but wireless Apple magic trackpad 2 works only when connected via USB .If I connect  via Bluetooth it detects as a mouse .
any progress on Bluetooth functionality .

AhmadChaiban, 2020-06-05:
Duplicate of #229

iomect, 2020-06-08:
Thank you very much for your contribution.
Just send 3 coffees for you~
Wait for MagicTrackpad 2 bluetooth support.

closed 2021-06-01 by imbushuo