imbushuo/mac-precision-touchpad #303: MacBook Pro 13" (2020) support?

FredegarBuckfast, 2020-05-20:
Hello there.

I just got a new MBP 13" (the 10th gen model, with 4 thunderbolt ports) and this driver doesn't seem to work. I install the inf, it tells me it succeeded, but nothing changes: device manager says nothing about a precision touchpad, and no gesture options are available in settings.

Am I doing something wrong, or is my model not supported yet?

Many thanks for this driver, I can't wait to get it working.

imbushuo, 2020-05-24:
PID/VID pair please?

FredegarBuckfast, 2020-05-24:
> PID/VID pair please?

Hi! According to the Mac's System Report, the PID is 0x027e and the VID is 0x05ac (Apple Inc.). Note that this is described as being for the 'Apple Internal Keyboard/Trackpad', so I'm not sure if you need a separate PID/VID pair just for the trackpad. If so, I wonder if you could tell me how to find this?

Many thanks.

imbushuo, 2020-05-26:
That's should be sufficient. Since T2 driver is pretty generic, the rest is just adding it into the INF file and I will do that.

---
Thanks,
Bingxing Wang

FredegarBuckfast, 2020-05-26:
Great, thanks!

closed 2020-05-26 by FredegarBuckfast