imbushuo/mac-precision-touchpad #305: click first finger and hold. touch second finger and move. random leaps happen and finishing even if first finger still on touchpad

segatrade, 2020-05-25:
- [yes] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

probably https://github.com/imbushuo/mac-precision-touchpad/issues/288 similiar

**Describe the bug**
click first finger and hold. touch second finger and move. random leaps happen and finishing even if first finger still on touchpad

**To Reproduce**
Steps to reproduce the behavior:
1)cmd+shift+s rectangular snip mode
2)click left side of touchpad 1 finger and hold
3)start move with second finger to select snip zone - sometimes I even cannot touch second finger - selecting already finished. Sometimes I can move part of zone, but one moment selecting finished or leap randomly.
* also I noticed that if I click first finger closer to touchpad center - selection is more stable. If I click closer to corner - it's less stable. But in MacOS it's always absolutely stable
* also in some games (ex. Civilization 4) it's uses right click + hold for choosing where unit to move. Similiar I start moves and it's finishes in random moment, but my first finger still on touchpad

**Expected behavior**
As is in MacOS - selecting snip zone finishes only when I release first finger. also there is no random leaps

**Traces**

**Environment**
Widnows 10 1909 (earlier versions worked the same)
Macbook Pro Retina Early 2013 15
Release 3805 Drivers-amd64-ReleaseMSSigned.zip (earlier versions worked the same)

omegamalkior, 2020-07-11:
Same. Really makes the experience a bit cumbersome as it happens almost all the time

rbreaves, 2020-12-23:
I sorta figured out what is happening here and this bug has existed for several months if not over a year. Basically ever since I used it - still a great driver, with this ONE flaw lol.

What seems to be happening and as I mentioned on the related post #288 is that it thinks you are double tapping while highlighting. IF it registers a double tap and you are not at the top of the document then it is as good as pressing the Home or PageUp button while highlighting and you will be sent to the top of the page or paged upwards by entire pages I think.

I might be wrong - but it feels like something other than rapid page acceleration even though that is what it feels like initially because you are witnessing the motion. It probably also depends on the app and your sensitivity settings - I have now moved mine to the lowest sensitivity setting and will report back later and how that is working out.

intxgo, 2021-08-14:
Hi, describing this in a simple words - lifting finger off the touchpad ends left click which is still being held.

Typical scenario: 
1. press win+shift+s
2. with left hand click and hold the low-left corner => left click
3. with right hand start selecting rectangular area on the screen
4. lift up your right hand finger to move across the touchpad to be able to repeat the move and extend the selection even further
5. lift up left hand finger => end of left click
6. selected snipped copied to the clipboard

When point 4 is not needed it's not a big deal, as you end up with the area you selected, but very often you need to move your finger several times across the touchpad to select all you need, which is currently impossible (but it works with stock Apple driver)

rbreaves, 2021-08-14:
Apple has released an official touch precision driver. You might want to try that?