imbushuo/mac-precision-touchpad #308: Mid 2020 Macbook Pro

kkx82960077, 2020-05-29:
I recently switched to the new Macbook pro released in Mid 2020.
What I found out is I can install the inf. file and it shows as the operation is successful.
However, nothing happens and I checked the device manager the driver is still apple multi-touch.
I am not sure whether it is because the driver does currently not support this such a new laptop or I faced the same issue met by the guy who get 2020 macbook air.
Thank u so much！！！

imbushuo, 2020-05-31:
Correlates #295.

xeonii, 2020-06-01:
Hey imbushuo,

Thanks for the great work. I'm struggling with the Macbook Pro 2020.

I've disabled secure boot, and I've gone ahead and performed "bcdedit -set TESTSIGNING ON" in cmd as administrator.

I downloaded everything and tried to install the inf file from my desktop in the following directory: 

_mac-precision-touchpad-master\mac-precision-touchpad-master\src\AmtPtpDeviceUniversalPkg_

It says operation completed successfully.

However, my device manager is still showing "Apple Multi-touch". I tried to manually install the .inf file but I get an error claiming it wasn't designed for x64 windows. 

Am I doing something wrong?

imbushuo, 2020-06-01:
@xeonii You are downloading source code (which obviously won't work.) You can click the green chechmark and download compiled binaries from AzDO artifacts (labeled with something like "5 published") and remember to choose the proper architecture.

I am holding off a release right now to resolve a few issues, then I will sign off a WHQL signed release so you can turn Secure Boot back on.

xeonii, 2020-06-01:
@imbushuo you're a lifesaver - it worked! 

For others: 
After downloading above, go to Device Manager and select HID > Apple USB Precision Device > Update driver and select the .inf from disk.

Thank you so much!

kkx82960077, 2020-06-02:
It works! Niceee! Thank u imbushuo.

imbushuo, 2020-06-13:
I am waiting for a HSM fix before I can submit it to MSFT for the final signing and release.

imbushuo, 2020-06-19:
See https://github.com/imbushuo/mac-precision-touchpad/releases

closed 2020-06-19 by imbushuo

zhoupeiyi123, 2020-09-26:
> @imbushuo您是救生员-奏效了！
> 
> 对于其他用户：
> 进行上述下载后，转到设备管理器，然后选择HID> Apple USB Precision设备>更新驱动程序，然后从磁盘中选择.inf。
> 
> 非常感谢！

Hello, can you help me, I am also mbp2020, but there is no response after installation