imbushuo/mac-precision-touchpad #310: Accidental finger touch caused when one finger tap and two finger scroll up/down

zprood, 2020-06-05:
**Pre-report checks**

- [1] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
Hi, I'm using trackpad 2 on Alienware laptop with Windows 10 2004. After installing, it works good via USB mode. But when I use the trackpad for browsering and normal work, faced some problems.
1. Tap to click, when I use single tap as mouse click, the cursor moves a little, can't focus the object such as links/small stickers, have to re-move the cursor and slight tap again. 

2. Two/Three finger scroll causes extra touch. This issue happens more frequently when I scroll page down/up on browser or document, will cause extra touch then trigger next level gesture. For example, I set three finger swipe UP for task view, Down for desktop, when using two finger scroll up on browser, it will trigger task view gesture. Similarly, scroll down, will trigger desktop gesture. I'm pretty sure that I just have two fingers on the trackpad not 3. 

**To Reproduce**
Tried reset the touchpad setting, event turn off the some gestures. 

**Expected behavior**
When single tap, the cursor will stop moving, no more extra touch happen because I use two finger scroll more frequently.

**Traces**
Don't know how to do that?.

**Environment**
Trackpad 2, Windows 10 v2004, USB user-mode.

omegamalkior, 2020-07-12:
First issue does not happen on mine. The second one you crossed out does happen a bit to me

DavidKarlas, 2020-11-15:
I also thought this was bug... After searching some more about trackpads and stuff I think I found what problem is...
[image]