imbushuo/mac-precision-touchpad #312: Trackpad stutters and lags

aayushbhatia06, 2020-06-07:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

// Yes I have 

**Describe the bug**
A clear and concise description of what the bug is.

// So I installed this driver and everything works as it should, the gestures the scrolling etc. The only issue is when I open a app (Say Microsoft Edge , Photoshop etc) or if anything is "loading" (i.e the loading cursors is visible) the pointer lags heavily almost like if it has shifted from 60 Fps to 10 Fps. This does not happen with all apps namely the ones that do not bring the "loading" cursor (File Explorer, Microsoft To-Do, VideoTape etc).

And this is an trackpad issue because I have tried five different external mice (Wired and Wireless) and none of them replicated this issue.

**To Reproduce**
Steps to reproduce the behavior: 

// Open any "heavy" application (Microsoft Edge, Photoshop etc)

**Expected behavior**
A clear and concise description of what you expected to happen.

The pointer should behave normally 

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.

// Macbook Air7,2 i.e Early 2015 13.3 inch
i5 5250u, 8GB memory and 128GB storage

imbushuo, 2020-06-13:
There is some discussion about moving existing UM driver into KM to address the problem, but is not yet being executed.

johnothwolo, 2020-06-23:
I'm also getting the same problem I think (MBP 13" 2019 ). At first it was a bit unresponsive, notably with scrolling and cursor speed. Thinking it was sensitivity, I just bumped up the settings, then I realized it's just unresponsive.

appy100, 2020-06-24:
> **Pre-report checks**
> 
> * [x]  Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?
> 
> // Yes I have
> 
> **Describe the bug**
> A clear and concise description of what the bug is.
> 
> // So I installed this driver and everything works as it should, the gestures the scrolling etc. The only issue is when I open a app (Say Microsoft Edge , Photoshop etc) or if anything is "loading" (i.e the loading cursors is visible) the pointer lags heavily almost like if it has shifted from 60 Fps to 10 Fps. This does not happen with all apps namely the ones that do not bring the "loading" cursor (File Explorer, Microsoft To-Do, VideoTape etc).
> 
> And this is an trackpad issue because I have tried five different external mice (Wired and Wireless) and none of them replicated this issue.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> // Open any "heavy" application (Microsoft Edge, Photoshop etc)
> 
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> The pointer should behave normally
> 
> **Environment**
> Add your device model, driver type (USB or SPI) and OS version here.
> 
> // Macbook Air7,2 i.e Early 2015 13.3 inch
> i5 5250u, 8GB memory and 128GB storage

Me too facing same issue having same model laptop

omegamalkior, 2020-07-12:
Does not happen to me, but considering I have a Windows laptop instead of a Macbook might be the reason

commit 04cb87d6 (2021-05-28): um: raise UMDF host priority to address input lag

Should resolve #312, #60 and as part of interim solution fo #370.
closed 2021-05-28 by imbushuo