imbushuo/mac-precision-touchpad #317: 3 fingers to drag

gchamort, 2020-07-05:
**First of all**
Well done man, amazing work not surprised its becoming the reference *geniun clap clap*

**Feature request**
It miss the feature of draging objects with the **3 finger drag** gesture you can find on macos as part of the "Accessibility feature" 
System Preferences > Accessibility  > Mouse & Trackpad > Trackpad Options > Enable dragging >> three finger drag

**Additional context**
See on apple support page : [Turn on "three finger drag" for your Force Touch trackpad](https://support.apple.com/en-us/HT204609)

**Miscellaneous**
That would be just amazing, feeback of a long time back to back "mac os on macbook and windows on tower" daily user it would be the cherry on top for an absolute perfect seemless experience
Thousand thanks, keep the good work

zhendosina, 2020-09-02:
yep, it have to be great!

roddymetanez, 2020-09-13:
This would be amazingly useful as it would enable panning and rotating in 3d programs.
Just learning to code, but count me if for beta tests, etc.

davecangit, 2020-09-16:
I would love for this too. This is one of the main trackpad functions I use. First setting to be changed when I log into a mac for the first time.

mattkefford, 2020-11-03:
I came here to request this feature too and glad to find it has already been suggested :)

Since I use macOS 75% of the time and BootCamp/Windows 25% of the time, whenever I am on Windows I find myself trying to three-finger drag things and being surprised when the windows suddenly change or disappear.

I understand that this is not simply interpreting the touch input on the trackpad itself but a much bigger task of writing a windows driver with multiple features, animations and probably a GUI for options, so probably not an easy thing to add. Given the amazing work you have done with this trackpad driver though I think you could do an awesome job of it and people would be prepared to pay a few dollars for it.

roddymetanez, 2020-11-04:
I can highly recommend Magic trackpad as a great utility that allows you to use the trackpad as a true three button mouse replacement.

mattkefford, 2020-11-05:
> I can highly recommend Magic trackpad as a great utility that allows you to use the trackpad as a true three button mouse replacement.
Hi Roddy, thanks for your suggestion but I was hoping to use the built in trackpad on my macbook rather than buy external hardware (especially when it costs £129!).

roddymetanez, 2020-11-05:
Apologies, I think at cross purposes. I meant these guys; magicutilities dot net<https://magicutilities.net/>

I think they do for MacBooks as well as the magic trackpad.

R

mattkefford, 2020-11-05:
Oh I see, this is an alternative driver foe the trackpad that runs on Windows? 

It doesn’t mention whether it works with built in trackpads but it does mention boot camp and there’s a free trial so no harm in trying it.

Foe the most part I am happy with the imbushuo driver though, it just needs a couple of tweaks to be perfect :) 

Regards, Matthew

XenorPLxx, 2021-04-28:
+1, that's the only thing that doesn't allow to switch my mouse with trackpad full time.

altosaar, 2021-04-29:
That would be awesome for me too! Magic trackpad ftw :)

daanta-real, 2021-06-02:
This is a MUST

MrJonhnny, 2021-06-10:
+++

Lawcheehung, 2021-06-10:
Agreed, this is my most missed feature from using the touchpad on MacOS!

imbushuo, 2021-06-12:
For everyone in this thread: open this feature request at Feedback hub. Feel free to post the feedback link here (as I might be able to help you triage it to the team.) It's a feature that needs to be implemented by Microsoft, so my driver can do little here.

closed 2021-06-12 by imbushuo

deluxghost, 2021-06-13:
some feedback links:

https://aka.ms/AAcw72b
https://aka.ms/AAcw72c
https://aka.ms/AAcw72d