imbushuo/mac-precision-touchpad #319: Phantom touches on Magic Trackpad 2

Swicked86, 2020-07-06:
In Win10 (1903) the touchpad is often registering an additional finger touching the pad.  Scrolling down in a page often closes the app.  It also seems related to the cursor jitter, and taps that I don't make .  Sensitivity has been reduced to it's lowest setting and still having these issues.  Perhaps the touch area is to small?

nemtsv, 2020-07-10:
This is how I can reproduce: swipe cursor with finger, when you release finger continue movement while hovering above surface ~3mm but not touching

omegamalkior, 2020-07-11:
Same!! Hope it gets fixed as it renders the touchpad fairly unstable when clicking after scrolling

hexagramg, 2020-08-11:
The same thing for me

adventures, 2020-12-12:
Any possibility to have this fixed? This is the only aspect of this driver that keeps me from using Windows 10, as the Magic Trackpad is the only device I can use due to hand strain. I'd be very grateful. :)

msavazzi, 2021-01-04:
Same for me. It explains why sometimes it perceives a double touch