imbushuo/mac-precision-touchpad #321: not working on 2020 macbook air?

zortrop, 2020-07-08:
Just did a clean Windows 10 install (boot camp) on 2020 MacBook Air. Installed the latest release of the amd64 driver. Precision trackpad doesn't show up in Device Manager. 

I also have a 2019 Air, and using the same steps it worked fine.

Thanks for this great driver!

imbushuo, 2020-07-11:
@zortrop Can you provide your device ID? I probably missed a few.

zortrop, 2020-07-14:
Yes I just emailed it to you. Thanks!

imbushuo, 2020-07-14:
@zortrop Ah, not that serial. Please send this one:

[image]

zortrop, 2020-07-14:
@imbushuo oh, here you go:

Apple Internal Keyboard / Trackpad:

  Product ID:	0x0280
  Vendor ID:	0x05ac (Apple Inc.)

USB2019, 2020-07-16:
So，can you release the newest version that support the 2020 macbook air? @imbushuo 

which is in https://github.com/imbushuo/mac-precision-touchpad/releases

imbushuo, 2020-07-16:
@USB2019 @zortrop I would like you guys try the artifact in the mentioned commit (which jumps to AzDO, remember to turn off Secure Boot) and see if that works or not. If so, I will sign a release.

USB2019, 2020-07-16:
Drivers-amd64-ReleaseSigned.zip  in AzDO  works perfectly!!! Thanks a lot. @imbushuo

zortrop, 2020-07-16:
sorry, i don't see the artifact / AzDO link? (new to github) thanks!

imbushuo, 2020-07-17:
@zortrop https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3834&view=artifacts&type=publishedArtifacts

zortrop, 2020-07-17:
Got it, thanks. Works perfectly, far as I can tell!

imbushuo, 2020-09-10:
Hey, check out https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2009-3835.