imbushuo/mac-precision-touchpad #325: Driver causes conflict with Laptop's built-in one

installgento0, 2020-07-13:
**Describe the bug**
Although the Magic Trackpad 2 works (albeit with a sensitivity issue as described in #237) as well as the built-in laptop's touchpad (in my case a Lenovo Yoga S740 with Microsoft precision drivers support oob), when the Magic Trackpad 2 device is disconnected from the USB port and turned off (as well as my laptop's bluetooth adapter), sometimes the built-in laptop trackpad misbehaves and registers inputs that were not made.
In order to make sure this wasn't my laptop's touchpad fault, I tried making sure the area on my touchpad was dry in order to be sure the issue wasn't being caused by humidity on the touchpad sensor which would falsely register inputs (kinda like what happens on certain smartphone displays) after that, the issue continued and then finally, I decided to uninstall the mac precision touchpad drivers by following the uninstall guide on the github page. After 2 days of usage without the drivers, I confirmed that the issue was no more.

**To Reproduce**
1 - Connect the Magic Trackpad 2 to the USB-A 3.0 port
2 - Turn on the Magic Trackpad 2 device
3 - Install the drivers by cloning the repo and right-clicking the file "AmtPtpDevice.inf" and "Install"
4 - Disconnect the Magic Trackpad and then, turn it off
5 - Use the built-in laptop trackpad and try using gestures and then leaving the trackpad alone
(sometimes, the trackpad will register inputs on it's on and spaz out for like a second).

**Environment**
Devices: Magic Trackpad 2, Lenovo Yoga S740
System: Windows 10 update 2004

omegamalkior, 2020-07-13:
Cannot reproduce with my HP Spectre x360 15 2020 with precision drivers. Have you tried doing this with other laptops to see if maybe it's just an incompatibility with your specific laptop?

imbushuo, 2020-07-14:
I guess this is a Windows issue, which is beyond my control (likely the Win32 state machine gets confused.)

13098766119, 2020-07-14:
大哥，有没有安装教程啊，我的是mac book air 2020，win10 企业版2004；求救，谢谢

installgento0, 2020-08-12:
> Cannot reproduce with my HP Spectre x360 15 2020 with precision drivers. Have you tried doing this with other laptops to see if maybe it's just an incompatibility with your specific laptop?

I will try this again with a Samsung laptop (synaptics touchpad) and report back eventually.

imbushuo, 2021-06-12:
Close this ticket since it is stale.

closed 2021-06-12 by imbushuo