imbushuo/mac-precision-touchpad #326: Total Commander delayed context menu isn't appear

SayusiAndo, 2020-07-16:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
No delayed context menu in Total Commander.
Delayed menu displays only when right-bottom corner is enabled and I have to "click" there.

**To Reproduce**
Install Total Commander
Disable right-bottom corner left click option
Start Total Commander
Navigate to a folder and double-finger right click on it for a few seconds and the context doesn't appear

**Expected behavior**
I expect the TCM context appear after 1 second

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
MacOS 2019 16'

imbushuo, 2020-07-17:
@SayusiAndo I actually have similar problem on Surface Pro X with a few applications including FDM. I think TCM is based on Qt too? In such case it is actually beyond my control so you should file bug with Microsoft (via Feedback Hub) as well as Qt community.

closed 2020-07-17 by imbushuo

SayusiAndo, 2020-07-17:
@imbushuo Thanks for the quick reply! The strange is that with Trackpad++ I didn't have this problem. Anyway, I'm kind of ok with it.