imbushuo/mac-precision-touchpad #328: Random BSODs on Windows 10 version 2004

enricmarques, 2020-07-18:
I have a MacBook 12" 2017. I've been using this driver since mid-2019 without any issue. Recently I've upgraded to Windows 10 version 2004, and I've been experimenting with some kernel crashes (BSOD).

At first, I thought it was a hardware issue, but as the same laptop is perfectly stable under macOS, I decided that one of the installed drivers wasn't compatible with the latest version of Windows 10. Then I enabled the "Driver Verifier" feature from Microsoft (https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/driver-verifier), enabling it for all installed drivers, and Windows 10 never boot again. Luckily I created a Restore Point before enabling "Driver Verifier" as recommended by Microsoft, so I rolled back Windows 10. After checking the file MEMORY.DMP with a dump viewer, the driver "AmtPtpDeviceSpiKm.sys" seems to be the problem. This might be related to the stability issues, or just this driver doesn't support the stress tests performed by "Driver Verifier".

Finally, I decided that my Windows 10 installation was corrupted, so I reinstalled Windows 10 (version 2004) from scratch, including this driver, and the BSOD happened again. So I rolled back to the stock Apple SPI Trackpad driver, and it seems BSODs are gone.

Has anyone experienced some stability issues with this driver in Windows 10 version 2004?

Thanks!

imbushuo, 2020-07-19:
I haven't run it under Driver Verifier. I do know my upper driver - Apple SPI device might also get involved, but I have not yet looked into that.

imbushuo, 2021-05-30:
Should fixed: https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2105-3965

closed 2021-05-30 by imbushuo