imbushuo/mac-precision-touchpad #330: macboo

hojin1767, 2020-07-19:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.

escomp9, 2020-07-24:
I have same experience. I have Macbook Air 2020 and installed Windows 10 version 2004 with bootcamp. This driver doesn't work. Let me know if you need more info. Cheers!

hojin1767, 2020-07-26:
did you fix this ???

hojin1767, 2020-07-26:
> I have same experience. I have Macbook Air 2020 and installed Windows 10 version 2004 with bootcamp. This driver doesn't work. Let me know if you need more info. Cheers!

did you fix this ???

omegamalkior, 2020-07-26:
> 
> 
> > I have same experience. I have Macbook Air 2020 and installed Windows 10 version 2004 with bootcamp. This driver doesn't work. Let me know if you need more info. Cheers!
> 
> did you fix this ???
I think the dev made a specific driver on one of these issue posts specifically made for the MBA 2020 but I'm not sure where it is rn. Search up one of the older posts and I think you might find it.

imbushuo, 2020-07-31:
@hojin1767 There is a special driver for MBA 2020 which is linked in another issue, it has not yet been signed for release.

imbushuo, 2020-07-31:
FYI it is https://github.com/imbushuo/mac-precision-touchpad/issues/321

escomp9, 2020-07-31:
> FYI it is #321

Thanks! I used the link in #321 and it seems to work fine. I had to turn off Secure Boot to make it work but after that it works great. Now bootcamp is finally usable thanks to this driver! Great work.

hojin1767, 2020-08-01:
> FYI it is #321
Thanks for your answer.
I already did it. but I didn't work in window 10 20H1 2004. 
I look forward to releasing your driver. thank you!

imbushuo, 2020-08-09:
@hojin1767 Have you tried to turn off Secure Boot?

hojin1767, 2020-08-09:
> @hojin1767 Have you tried to turn off Secure Boot?

your driver works well when I turn off Secure Boot. thank you!!! 👍 
but if I turn on Secure boot again, your driver didn't work. so, mouse disappear.
if I want to use  your driver, should I always turn off secure boot?

imbushuo, 2020-08-13:
@hojin1767 I am going to sign a release shortly so you can leave Secure Boot on.

hojin1767, 2020-08-14:
> @hojin1767 I am going to sign a release shortly so you can leave Secure Boot on.

I really appreciate your help 👍 👍

imbushuo, 2020-08-25:
@nuvopm Unfortunately I am in the process of rotating code signing certificate. The verification process might takes a few more days to finish.

sun47451685, 2020-08-30:
> @nuvopm Unfortunately I am in the process of rotating code signing certificate. The verification process might takes a few more days to finish.

我是雷蛇灵刃15用户，在最近的windows升级后也遇到了类似的问题，期待新版的解决。

nuvopm, 2020-09-06:
> @nuvopm Unfortunately I am in the process of rotating code signing certificate. The verification process might takes a few more days to finish.

Any updates on this? Been waiting for the new release, thanks!

imbushuo, 2020-09-06:
Certificate token shipped on Friday, we need a few more days to provision this.

nuvopm, 2020-10-29:
> Certificate token shipped on Friday, we need a few more days to provision this. Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

Do we have any updates on this? Thank you.

imbushuo, 2020-10-29:
We actually shipped a version after then. Have you yet tried that one?

---
Thanks,
Bingxing Wang

________________________________ From: nuvopm notifications@github.com<mailto:notifications@github.com> Sent: Sunday, September 6, 2020 10:41:35
AM To: imbushuo/mac-precision-touchpad mac-precision-touchpad@noreply.github.com<mailto:mac-precision-touchpad@noreply.github.com> Cc: Bingxing Wang i@imbushuo.net<mailto:

nuvopm, 2020-10-30:
> We actually shipped a version after then. Have you yet tried that one?
> […](#)
> --- Thanks, Bingxing Wang
> ________________________________ From: nuvopm <notifications@github.com> Sent: Thursday, [...]

@imbushuo 
Didn't notice there was a new release. It works flawlessly on my MBA 2020, many thanks!
However, is there any way to increase the cursor speed? As the max speed on the trackpad settings is still too slow for me.