imbushuo/mac-precision-touchpad #333: request 3-finger drag feature.

ShinChven, 2020-07-27:
Thanks for your work, you did a great job to port the driver to Windows.

Hope it implements the 3-finger drag feature.

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Implements the 3-finger drag feature.

**Additional context**
Please search `3 finger drag` in [this page](https://support.apple.com/en-us/HT204895), if you haven't use it before.

imbushuo, 2020-07-31:
Tracked in https://aka.ms/AA92oyn

closed 2020-07-31 by imbushuo

VICLER, 2021-01-31:
Ist this feature now implemented? 
That is only one thing that I miss in this wonderful Driver.

ShinChven, 2021-02-01:
> Ist this feature now implemented?
> That is only one thing that I miss in this wonderful Driver.

Last week I checked: no.

VICLER, 2021-02-10:
@imbushuo please implement this feature. With 3 finger drag I can completely replace my mouse with a trackpad. It is very usefull and comfortable! Thank you very much!