imbushuo/mac-precision-touchpad #342: Error installing on windows 10

Coon1123, 2020-09-03:
**Describe the bug**
When i plug the MT2 via USB cable, the system recognizes a "HID-Compliant Mouse". Then, when I try installing the driver on the Device Manager, I get the error "...make sure it is designed to work with windows for x64-based systems". I am trying to install AmtPtpDevice.inf that I decompressed from Drivers-amd64-ReleaseMSSigned, release 3826.

**To Reproduce**
See above. 

**Expected behavior**
I expected the driver to install, windows to recognize the MT2 as a trackpad and give it funcionality as it does its synaptics trackpad. Right now i can only move the cursor and click, cant use more than 1 finger on it.

**Traces**
Dont know what WPP tracers are...

**Environment**
Lenovo Thinkpad T450, with windows 10 enterprise, version 1809, built 17763.1339
[image]

Bananaman, 2020-09-11:
Install chocolatey: https://chocolatey.org/

Then open a powershell admin prompt and type: `choco install mac-precision-touchpad` (it installs this package: https://chocolatey.org/packages/mac-precision-touchpad/)

That's all. To update in the future you go to a powershell admin prompt and type `choco upgrade all`

niklase2, 2021-03-07:
I hope I'm not too late, but you probably downloaded the wrong version for your computer. When you choose which version you want to download, there are 2 options: **arm64 and amd64**. In videos or instructions you are going to see the amd64 one, because it is the right choice for most people, but maybe you have to choose the other one (arm64). Also make sure that you first unzip it and **RIGHT CLICK THE 2ND FILE** in the folder and install.
Has this solved your issue?

imbushuo, 2021-06-12:
Closed since this is a stale ticket.

closed 2021-06-12 by imbushuo