imbushuo/mac-precision-touchpad #343: TODO: Reprovision EV cert [WIP]

imbushuo, 2020-09-08:
Release task has been stuck for a while until this completes. Time estimation is this week or next week.

zprood, 2020-09-09:
Thanks a lot, can't wait for new release.

Bananaman, 2020-09-10:
@imbushuo Speaking about certificate...

README.md has old information. It talks about manually installing a certificate and links to https://github.com/imbushuo/mac-precision-touchpad/issues/228 instructions. But I checked the package. You do not have a manual certificate. You have a real Microsoft certificate. So you can delete that information from the README. :)

imbushuo, 2020-09-10:
Yes, we ramped up MS cert recently, but I didn't sync the README. That line will be removed in the future.

---
Thanks,
Bingxing Wang

closed 2020-10-18 by imbushuo