imbushuo/mac-precision-touchpad #345: Question: Does the Magic Trackpad 2 have firmware revisions?

Bananaman, 2020-09-11:
This is a question for @imbushuo and everyone else.

I have been Googling, and there aren't any straight answers. But I found a hint that there is a hidden firmware updater in macOS which will send new firmwares to the Magic Trackpad 2. Someone had asked about troubles with their trackpad and posted a log at https://apple.stackexchange.com/questions/321948/why-does-my-magic-trackpad-2-randomly-disconnect-and-stop-clicking. The log contains the following words:

> Got device attach event for com.apple.mau.plugin.standalonehiddevice.trackpad.6.bt updateInProgress=0 queue[0]=(
mobileassetd    handleCopyMatchingAssets: HANDLER - assetQuery [com.apple.MobileAsset.MobileAccessoryUpdate.TrackpadFirmware.6]:

So it seems to me like Apple secretly updates the Magic Trackpad 2 firmware periodically.

The questions are:

- Does having the latest firmware matter for using your driver, imbushuo?
- Is there any way to do the firmware updates on Windows or will I have to borrow a Mac?
- If I borrow a Mac, do you think it will need to have the latest macOS to get the latest trackpad firmware? Or do you think it would be enough to have any version that contains the "com.apple.MobileAsset.MobileAccessoryUpdate.TrackpadFirmware" code? (I suspect that the firmware updater is enough, and that it will grab the latest firmware no matter what macOS version the owner has.)

imbushuo, 2020-09-11:
> Does having the latest firmware matter for using your driver, imbushuo?
I don't think so. Also, I don't see the point of constantly changing firmware and driver (since the input device is supposed to maintain some compatibility.)

> Is there any way to do the firmware updates on Windows or will I have to borrow a Mac?
Probably since it is just a USB/Bluetooth device. But this can lead to copyright concerns.

> If I borrow a Mac, do you think it will need to have the latest macOS to get the latest trackpad firmware?
I don't know.

---
Thanks,
Bingxing Wang

Bananaman, 2020-09-11:
Good point. Their firmwares are probably never going to change the communication protocol that the driver uses since it has to be compatible with older Macs too. This also suggests that the latest macOS won't be needed to get the latest firmware, but can't know for sure. So having outdated firmware should be fine.

As for windows updates of firmware, it would be a lot of extra work for you, so I am not suggesting you do anything. You'd have to find out how their updater checks the firmware version online, how it downloads it, and how it installs it safely. It's just a lot of work for very little benefit, since old firmwares probably don't matter, as we've established. :-)

I did have an idea about how to update from Windows though. I'll get VMware Player (free virtual machine) and do the "Mac virtual machine on Windows" trick (there are guides about how to use VMware player for that), with the latest macOS version, and then connect the trackpad to the virtual machine (by giving it complete control of the USB port) to just let it update that way. This is my suggestion to anyone else who wants a firmware update without having to find a Mac. The reason I'm gonna update it is because my firmware is definitely from 2015/2016 since it hasn't been connected to a Mac since then. And the website I linked to above talks about firmware updates more recently than that. So the updates might make it perform a bit better and would be worth it. :-P

Bananaman, 2020-10-02:
Closing this since it has been answered.

closed 2020-10-02 by Bananaman

WinkelCode, 2023-01-02:
FYI, Firmware seems to have caused and solved this issue: https://github.com/imbushuo/mac-precision-touchpad/issues/477

gingerbeardman, 2024-01-29:
OTA Magic Trackpad firmware updates are listed here: https://appledb.dev/device-selection/Trackpad.html