imbushuo/mac-precision-touchpad #346: driver stability issues

viper2345678901, 2020-09-11:
I'm using the latest one with the magic trackpad 2, but the driver works and then stops working sometimes and then I get an error saying that the usb device is not functioning properly.

TylerMSFT, 2020-09-22:
I've hit this too. It got much worse when I upgraded to Windows version 20HW (19042.508)
It happens multiple times a day. If I simply turn the trackpad off/on, it starts working again-for awhile.

TylerMSFT, 2020-09-30:
This is definitely related to the Windows 20HW update. I downgraded my OS to the previous version, and the driver worked well. I was upgraded to 20HW again (work requirement) and the driver started failing multiple times an hour.

TylerMSFT, 2020-10-09:
I updated to Windows 20H2, today, and the driver is working better - I don't have to keep turning the trackpad on and off to reset.

imbushuo, 2021-06-12:
@TylerMSFT @viper2345678901 It might be related to the custom USB HID transport I implemented a few years ago. I am working on moving to system HID transport stack now (no ETA yet, Bluetooth is on it though.) This involves a hack with WDM and the way HID stack works, I hope they don't change anything soon.

TylerMSFT, 2021-06-14:
Thank you, @imbushuo.  The problem is back, intermittently. What I see is periodic crazy high system interrupt usage in task manager (46% of an 8 core i7 Surface Book 3)
The moment I unplug the trackpad, the system interrupt zero goes back to near zero.
Now I'm running Windows version 21H1, and see the same thing. It's not all the time, but probably 2-3 times a week. I haven't isolated what causes it. Maybe wake from sleep is involved.
Aside from this, I love this driver and how it allows me to use my trackpad with my Windows machine. I use it 8 hours a day and most weekends. Thank you very much for making it possible for me to use this trackpad! I look forward to seeing how your changes work out.

imbushuo, 2021-06-15:
@TylerMSFT Next time if you encountered high interrupt usage, please help me capture a WPR ETL trace:

[image]

This should help me better understand what happens, since I am unable to reproduce this locally (for a really long time.)

TylerMSFT, 2021-06-15:
Will do