imbushuo/mac-precision-touchpad #349: MacBook air 2013 安装出错，鼠标指针丢失

dfyun520, 2020-09-19:
你好，非常感谢你写了这么好的软件。
我的电脑是MacBook air 2013，安装win10，直接安装软件出错，鼠标指针丢失。
我此前并未安装过trackpad++。
另外，不知道会不会是这个原因：我更换了nvme硬盘，进入win的时候无法打开boot camp控制面板，虽然我已经按照网上的方法已经打开了，我不知道是否是这个原因导致安装不成功。
希望得到你的回答，非常感谢。

solensolen, 2020-09-26:
Please use English (as the author's language). I don't think he will understand your issue in other language. Even if he understands, he don't have to.

You have changed the hardweare, change the system(illeagal), facing some very special problem. I don't think your issue will be worth any other's time to solve. Especially you **dear to** flag this as bug, it's a humiliation to the author and yourself.

As a amateur, Please first lookup GitHub Community Guidelines. There's a lot things for you to imporve. 

Can someone **close this issue**? As one speaks and writes Chinease,  it's painful to see such g****e content presenting in Chinese.

imbushuo, 2020-10-18:
@dfyun520 Try to use an alternate input device (like a USB mouse) to check with Device Manager. I would like to know the error information in Device Manager.

liuziwendave, 2020-11-09:
> Please use English (as the author's language). I don't think he will understand your issue in other language. Even if he understands, he don't have to.
>
> You have changed the hardweare, change the [...]

No cool, man. So racist. Are there any regulations, guidelines, rules, or laws forbidden use other languages to express errors? if so, please raise one to me. And are you the author? or can you represent the author? why you say this humiliate author? are you trying to humiliate the author? 
By the way, this question still not closed :). From what I see, you are a little racist freak.

imbushuo, 2020-11-09:
I am aware certain MacBook Air 2013 has this issue, unfortunately I need to get a device to see what happened there, which will not happen very soon (so I leave this open.)

I do would like to have people sending issues in English (so others can search existing things without the language barrier), but I am okay to have other languages including Chinese (which I can understand.)