imbushuo/mac-precision-touchpad #350: [Misc] Looking for a full-time job

imbushuo, 2020-09-19:
I am almost going to graduate at this moment. Given the uncertainty at LinkedIn's intern return FTE offer at this moment, I decided to open an issue here for supporting my hobbyist projects in the future.

You can find my resume at [here](https://www.linkedin.com/imbushuo). I am comfortable with SWE, SRE and Firmware Engineering jobs, and preferably near the Bay Area.

evancharlton, 2020-09-26:
Just FYI: that URL is a 404 -- I think you meant to go to: `https://www.linkedin.com/in/imbushuo` 😁

imbushuo, 2020-09-26:
> Just FYI: that URL is a 404 -- I think you meant to go to: `https://www.linkedin.com/in/imbushuo` 😁 

Corrected. Actually at this moment I've got a few opportunities, but I will leave this issue here for a while.

closed 2021-06-04 by imbushuo