imbushuo/mac-precision-touchpad #353: Features like "Better And Better" APP

murphyzhao, 2020-09-26:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

Is [this](https://www.better365.cn/bab2.html) possible on Windows?

[image]

Bananaman, 2020-10-02:
Yes, using https://www.multiswipe.com any multi-touch input can support gestures.

You can close this ticket. :)

greeneryAO, 2020-10-04:
> Yes, using https://www.multiswipe.com any multi-touch input can support gestures.
> 
> You can close this ticket. :)

1. multiswipe actually let you buy it, there's known issue of dev not sending licence/supporting

2. multiswipe works with your precision drivers? did you have to do anything special in signing or any of that, all I did was click install dll of latest build, multiswipe won't recognize my gestures.

Bananaman, 2020-10-05:
@greeneryAO Hmm. I don't use MultiSwipe. Here's a free alternative which does the same thing:

https://gesturesign.win/

But I don't know if they have trouble detecting the multitouch precision touchpad. I haven't tried them. Perhaps @imbushuo knows.

greeneryAO, 2020-10-05:
> @greeneryAO Hmm. I don't use MultiSwipe. Here's a free alternative which does the same thing:
> 
> https://gesturesign.win/
> 
> But I don't know if they have trouble detecting the multitouch precision touchpad. I haven't tried them. Perhaps @imbushuo knows.

I tried emailing him yesterday, I think he is busy, gesturesign isn't quite as good because it's missing some features of multiswipe like left or right tiptap, but neither work for my macbook so it doesn't matter, if you don't mind, if you could download gesturesign and try I would appreciate, then I would know it's my macbook or not and if it is I'll buy a magic trackpad, thanks.

Etrit1, 2021-01-01:
Wait, did someone get MultiSwipe or Gesture sign working?

Neither are seeing my touchpad on a Late 2012 MacBook Pro.

Etrit1, 2021-01-06:
MultiSwipe seems to be dead.  The website is broken and you can't sign up to download a trial.

But that really isn't important because MultiSwipe etc will not recognize our touchpads.  (I tried MultiSwipe's test tool.. and no go, same with GestureStudio or whatever it is.)

However, since this dev got Windows to recognize it, you MIGHT be able to get SOME of the stuff BetterTouchtool gives you.  It's working for me on a Late 2012 MacBook Pro Retina.

Check out my long post on this issue thread.. 

https://github.com/imbushuo/mac-precision-touchpad/issues/355

imbushuo, 2021-06-12:
I think this is large enough to have another project. Since this is out of scope for the driver itself, close as of now.

closed 2021-06-12 by imbushuo