imbushuo/mac-precision-touchpad #355: gesture programs don't recognize on macbook air 2012? multiswipe/gesturesign, anyway to get them to pick up inputs?

greeneryAO, 2020-10-03:
my windows system sees the precision drivers and they work well, but those programs won't respond to the touchpad even though windows thinks I have precision touchpad, no way I'm touching trackpad++ again to try with that it, it disabled my touchpad i had to reinstall windows.

Etrit1, 2021-01-01:
Oh sigh to this not working with Multiswipe.

It's still better then the BootCamp Apple drivers.  My right click seems SO much more reliable then with either TrackPad++ or BootCamp.

As for touching TrackPad++ again, well there is no point with these drivers.   TrackPad doesn't let you customize much more then Windows does.

I want something that lets me do keyboard shortcuts like BetterTouchTool in Mac OS but oh well.

greeneryAO, 2021-01-01:
I spoke to the dev of the precision driver about this, he is busy but he said it will be difficult to get mac touchpads recognized by gesture programs, so it's unlikely.

I can't even get multiswipe on a windows laptop to work, well, it works, but I can't register it and trial runs out, seems to be a common complaint that it was abandoned, when did you register yours Etrit1?

Windows needs a btt badly.

Etrit1, 2021-01-06:
WOOT WOOT!  I figured it out!

Windows 10  sometimes DOES let you assign almost whatever Touchpad/Trackpad gestures you want. (I say sometimes because I'm not sure this will work on any all computers)

I'm on a MacBook Pro Late 2012 Retina

Windows 10   is showing something called "Advanced Touchpad Gestures" when I search for stuff in the StartMenu.  However, it just opens to the Metro Settings paged and I couldn't find any such Gesture settings easily in the options there.

Happily, I figured out a simple workaround.

Open the normal Touchpad gestures setting page.  Then click on "Advanced gesture configuration"  in the Related Settings to the right of that window.    

Then customize to your hearts content!. You yes can even do keyboard shortcuts!

Also, since Alt-Tab/ Task View on Windows 10 is such a piece of junk by showing Minimized Windows with no way to turn that off.. install the free BetterDesktopTool program.  Set whatever keyboard shortcut you want for its MANY options  (I just care about "Show non Minimized Windows",  and make a touchpad gesture for that command.

DONE!

Also, To fix the stupid weird Bluetooth Signal issues:  

I've looked around, some say check a box for your Wifi card in Device Manager that tells it to play nice with Bluetooth.  That didn't help on a 2.4ghz network connection.     I guess it did for others.    Others said do tons of annoying stuff.  I haven't tired that but I found some people saying it's an issue  with 2.4ghz wifi interfering with Bluetooth...

 so I switched to the 5ghz Wifi connection and ...  Problem fixed!!

It means I'm gonna have to buy a external network card for my entertainment laptop that only has 2.4ghz that I VNC to all the time but, oh well.  I at least now don't have to deal with Apple any more.

I MIGHT figure out how to remap keys on my keyboard idk.. I'd rather have Cmd be the most used key instead of Ctrl.. but that is a later project.

greeneryAO, 2021-01-06:
thanks for the post, I'll check out some of the things you mentioned.

closed 2021-06-12 by imbushuo