imbushuo/mac-precision-touchpad #357: looking for a fork to work with gesturesign and multiswipe gesture programs, can paypal.

greeneryAO, 2020-10-05:
the windows precisions swipe/etc options are nice but not being able to customize makes it not as useful as a mac using bettertouchtool, I don't know if this is only my system problem, macbook air 2012, maybe a magic trackpad would work, but it's expensive and more for desktop, would prefer my own touchpad to work with these programs, anyway, leave me email if interested in making the fork.

imbushuo, 2020-10-18:
@greeneryAO that is large enough to be implemented as a separate project imo. Technically this is a task for Microsoft, since the PTP input stack handles all gestures internally.

Etrit1, 2021-01-06:
See my very long comment on this issue thread.. you might be able to customize some 3 and 4 finger gestures.   It's not as good as BetterTouchTool for Mac OS but it is good enough for me.

https://github.com/imbushuo/mac-precision-touchpad/issues/355

imbushuo, 2021-06-12:
Close as of now, since it's an architectural issue which cannot be easily workarounded at my side.

closed 2021-06-12 by imbushuo