imbushuo/mac-precision-touchpad #358: Request to work with Magic Trackpad 1

meeebee, 2020-10-11:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?
Yes, does not exist.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Would like to see this driver updated to work with Magic Trackpad version 1

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Would like to see the same driver used for the first version of Apple's Magic Trackpad 1.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
Would probably be dependent on bluetooth support as Trackpad1 does not have USB option.

povlhp, 2020-10-29:
I would like Magic TrackPad 1 support as well. 
Can not use this driver.
I uses generic mouse driver. I can move and click. No right.click even with modifiers

fabeit, 2020-11-11:
There are only paid alternatives for TP1

lothar-cell, 2021-02-07:
I'd be also happy to see support for Magic Trackpad 1!

GenieTim, 2021-06-02:
Can't you just use the Bootcamp Drivers for Magic Trackpad 1? 
https://support.apple.com/downloads/bootcamp-windows-drivers -> download Boot Camp Support Software -> Unpack Zip -> Install the drivers needed from within the zip, such as zip:BootCamp/Drivers/Apple/AppleWirelessTrackpad64.exe and AppleMultiTouchTrackPadInstaller64.exe

lothar-cell, 2021-06-02:
Yes, that's what I use, but the right-click implementation is a pain, it misinterprets the taps.

MrJonhnny, 2021-06-10:
+++ to mt1 support

imbushuo, 2021-06-12:
[image]

🤔 Sounds not too bad, will be added in the backlog.

AFuFred, 2021-06-20:
That‘s great！
Thank YOU！

> [image]
> 
> 🤔 Sounds not too bad, will be added in the backlog.

gljones2001, 2022-01-16:
So does this driver have Magic Trackpad 1 support or no? Its not clear from this thread as to weather this has been implemented since the last comment or not.

CottonEvoker, 2022-02-02:
I'd also **really** like to see this made to work with Magic Trackpad 1.

alimas7, 2022-02-09:
> Can't you just use the Bootcamp Drivers for Magic Trackpad 1? https://support.apple.com/downloads/bootcamp-windows-drivers -> download Boot Camp Support Software -> Unpack Zip -> Install the drivers [...]

Built-in MacBook Touchpad works nicely with https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2105-3979.

Wireless Magic Trackpad 1 works **almost** as well with Bootcamp AppleWirelessTrackpad64.exe and the following registry flips in HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\HID\<find your VID PID from \Device Parameters per https://apple.stackexchange.com/questions/63598/invert-scrolling-on-windows-bootcamp/312504#312504.

```
"FlipFlopWheel"=dword:00000001
"FlipFlopHScroll"=dword:00000001
```

Disabling and reenabling the device is quick to get those registry changes to become effective, no need to reboot.

The mention of "**almost**" is because the Magic TP1 feels a bit jerky and doesn't behave as the built-in Touchpad.

For example, with the built-in Touchpad, I can swipe (two fingers swipe and let go) to scroll up/down and scrolling will slowly decelerate after fingers are lifted.  With the Bootcamp driver, swiping in the same manner abruptly stops as soon as fingers are lifted up from Magic TP1, no slow deceleration, doesn't feel "natural".

lothar-cell, 2022-02-09:
Also click and drag is interpreted as two-finger click, unless a pre-defined left or right lower corner is used for clicking. This and no proper “rubberband” scrolling (or touch gestures) makes a restrained user experience.

fabeit, 2022-02-09:
The other problem with the Boot Camp drivers is that they don't have any other gestures such as three finger drag, pinch and zoom, three and four finger gestures.

lothar-cell, 2022-02-09:
100% agree, and payed alternatives capitalise this situation (even extramagic, which is a half-baked solution anyway, as for magic utilities I despise subscription-based software.)

krackers, 2022-02-13:
@imbushuo Here's finger data frame format for MT1, based on both linux driver as well as my own reverse engineering.

```
struct __attribute__((__packed__)) mt1_finger {
	SInt16 X: 13;
	SInt16 Y: 13;
        UInt8 Unk1: 3;
	UInt8 Unk2: 3;
	UInt8 Touch_Major: 8;
	UInt8 Touch_Minor: 8;
	UInt8 Size: 6;
	UInt8 TrackingId: 4;
	UInt8 Orientation: 6;
 	UInt8 FingerId: 4;
	UInt8 HoverState: 4;
};
```

Still not sure what `unk1` and  `unk2` are though. They seem to be some sort of relative coordinates as `unk2` increases as you drag finger right and up, and decreases as you go down and left. `unk1` is similar but more sensitive.

krackers, 2022-02-20:
For completeness, here is header format

```
struct __attribute__((__packed__)) HEADER {
	UInt8 report_id: 8;
	UInt8 button_data: 6;
	UInt32 timestamp: 18;
};
```

Each frame consists of 4 byte header as above followed by 9 bytes of finger data as in other struct.. Finally to enable multitouch mode you have to send a report to the trackpad, I think you can get that from the linux driver. Let me know if any other information is needed here, but I believe this should be sufficient to begin reading frames from the trackpad? There are other set/get reports that are sent during the initial handshake though, I don't know how to decode those yet. I'm guessing based on MT2 work that some probably include trackpad dimensions, but this should be a fixed constant so it's not really needed.

GoldenJoe, 2022-02-25:
+1, it would add a lot of utility to the older model.

gljones2001, 2022-02-25:
All for the Magic Tracpad 1 unite!
And down with Magic Utilities!
DOWN WITH STUPID SUBSCRIPTION SOFTWARE!!!!

bruder-grimm, 2022-05-01:
A new trackpad would be more expensive than donating - I'd be happy to implicitly pay for MT1 support!

gljones2001, 2022-05-01:
I would too, but it ended up being used on a mac and I cannot send it now.  :(

fschuh, 2022-05-13:
+1 for Magic Trackpad 1 support! 
Please make it happen, lots of us still have some of these lying around :)

efouladi, 2022-09-24:
+1 and thanks for the awesome work

piereligio, 2022-11-20:
Is there any hope to get this supported? It would be amazing

superluig164, 2023-09-29:
I'd love to see this, recently got a Magic Trackpad 1 from a donation event and I'd love to be able to use it to its fullest on Windows.

CottonEvoker, 2023-09-30:
I'd still like to see this too @superluig164 , but I think we need to keep expectations in check at this point.

Prior to your post  it was 10+ months since this request last got a +1.

superluig164, 2023-09-30:
I think it's better to add my opinion and hopefully one day it happens than to let the post get even more abandoned and give the impression that nobody is interested in the feature anymore.  I'm not expecting it to happen tomorrow.  Just adding my +1 to help.

Mikry13, 2023-11-04:
+1 from me
Actually, I would like to research into the possible solution, but don't quite know where to start, because I'm completely new to Drivers/Bluetooth stuff. For now I'm collecting some logs for Magic Trackpad 1 with Microsoft Bluetooth Test Platform package and learning about Bluetooth reverse engineering as I have some free time on the weekends.

Maybe if someone with some knowledge is interested and we can communicate somehow (discord?) to slowly move towards the solution?

I guess it will be better than just waiting something to happen.

subjeti, 2024-01-14:
+1 

Would love to see the MT1 working on it's full glory on windows :)

Darthagnon, 2024-05-27:
I would love to see the Magic Touchpad 1 (A1339) working with precision drivers on Windows. I love that it has removable batteries, rather than a built-in batteries of the later models - makes it longer lasting, less disposable tech.

axeligl, 2024-06-13:
Hi! I got my Magic Trackpad 1 (A1339) working okey with click, right click, scroll... but I want to use the three fingers gesture to change desktop on windows 11. I got working with the bootcamp driver as @krackers  said. Is there anyway toget those gestures work?

GraphicHealer, 2024-06-14:
> @axeligl: You might try the [Magic Utilities](https://magicutilities.net/magic-trackpad/features) which supports the Magic Trackpad 1 with 3 finger dragging, even with drag-lock as seen in macOS.

Yeah, but the point of this github was to have an _open source and free_ alternative to Magic Utilities. They charge for their driver.

heyitzrare, 2024-07-18:
+1 for MT1 support - I found one CIB at a thrift store for super cheap, and I would _love_ to use it in my workflow :)

sjain882, 2025-06-28:
Would also love to see this!

dmitry-entin, 2026-10-29:
Draft PR with Magic Trackpad 1 (Bluetooth) support: #579

Heads-up: it was written with an AI assistant and tested on one computer only (Windows 11 x64, MT1 over Bluetooth). Cursor, tap, scroll, pinch to zoom and 3/4-finger gestures work there. It is not officially signed, so trying it needs a self-built driver and test signing enabled (which requires Secure Boot off).

If you try it, please report your results in the PR, especially on other hardware.