imbushuo/mac-precision-touchpad #362: Windows PC Installation Magic Trackpad 2

mr-vito, 2020-10-27:
**Pre-report checks**

- [Yes ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
I am trying to install Magic Trackpad 2 on my Windows PC. I am not using a MacBook/Mini nor I have installed BootCamp.
Just plain old Windows 10. I clicked on Install and I an operation complete message. The Magic Trackpad 2 does not change its behavior (no scroll, no pinch to zoom, etc). I tried both using via Bluetooth and via USB.

**To Reproduce**
Use a PC with Windows 10.
Connect the trackpad via USB or Bluetooth.
Download the latest release.
Right-click Install.
Operation Complete.

**Expected behavior**
The trackpad should also show up under Mouse in general Settings.
Zoom and scrolling should be working.

**Traces**
N/A

**Environment**
Windows 10 1806. Intel-based PC x64.
Magic Trackpad 2 (works fine with iPad Pro).

michaelgosling, 2020-11-05:
Installs perfect for me. What process did you follow for installation?

mr-vito, 2020-11-05:
It was actually the cable.

Some cables are only for charging, which was the my case.

I swapped the cable for one that could transfer data and the trackpad "magically" worked. The sound of the driver was also a giveaway...

Also, any updates on Bluetooth support?

Thanks!

vmo-mgosling, 2020-11-05:
I'm just a user but the ticket for tracking BT support is here: https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/634/

michaelgosling, 2020-11-05:
I'm just a user but the ticket for tracking BT support is here: https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/634/

imbushuo, 2020-11-09:
We currently track Bluetooth in another issue thread.

closed 2020-11-09 by imbushuo