imbushuo/mac-precision-touchpad #364: Click with left finger and drag with right finger to select an area

mattkefford, 2020-11-03:
I'm always frustrated when...I use snipping tool to select a portion of the screen and once I have started the selection, I cannot lift my right finger off the trackpad to adjust the selection. This is regarding the rectangular selection mode. As soon as you lift your right finger the selection is made and it also starts drawing/marking-up the selection in a way that cannot be deselected.

What would be great is if I could click using a finger on my left hand to start the selection, then use a finger on my right hand to drag the opposite corner of the selection box, be able to lift my right finger off the trackpad and place it again to fine tune the selection size, continue to drag the corner of the selection box. Only when lifting my left finger would the selection be made.

My workaround at the moment is to start making the selection with my right finger all the way in one of the trackpad corners, then quickly make the selection and lift both fingers. It's pretty awkward and usually takes a few tries to get the correct selection area.

Here is an example of me trying to select this text box, running out of trackpad, attempting to adjust but the snip is taken at that point and then for some reason drawing over the new snip with red pen:
[image]

michaelgosling, 2020-11-05:
Not sure what trackpad you're using but this doesn't happen on Magic Trackpad 2. Kinda sounds like default apple driver behaviour.

mattkefford, 2020-11-05:
> Not sure what trackpad you're using but this doesn't happen on Magic Trackpad 2. Kinda sounds like default apple driver behaviour.

I am using a Macbook Pro 13" (mid-2014) and the built-in trackpad. 
Do you mean that "sounds like default apple driver behaviour" when it ends the selection as soon as you lift your right finger? Because I can assure you that on macOS you can lift and adjust the selection area as much as you like whilst the left finger is still pressed down.
If you like I could record a short video to show the issue? Maybe I haven't described it well enough.