imbushuo/mac-precision-touchpad #368: Right click(two-finger click) and drag the mouse have a serious bug

Etern213, 2020-11-15:
**Pre-report checks**

Yes, but I found nothing to solve this problem.

**Describe the bug**
**two-finger click(Right click) and drag the finger have a serious bug.** When I use one-finger click(Left click) and drag the finger, its work normally. But when I use **two-finger click(Right click) and drag the finger**, its works the same as **one-finger click(Left click) and drag the finger**

**To Reproduce**
Use two-finger click and drag the finger to reproduce the problem. For example, in world of warcraft, Right click and drag the mouse is very commonly used.

**Expected behavior**
I think when I use two-finger click and drag the finger, it should works the same as Right click and drag the mouse.

**Traces**

**Environment**
I use macbook pro(ratina 13-inch early 2015), and install win10 with bootcamp.

Etern213, 2020-11-15:
Maybe my description is not clear enough, but I think reproduce this problem is very easy. Before I install it, two-finger click and drag is normal, but after I install it, some bug occurred. @imbushuo

funtikar, 2020-11-30:
you mean it only happens in WoW?

Etern213, 2020-12-02:
> you mean it only happens in WoW?

no, it happens all the time. I just ... use wow as an example.

niklase2, 2021-03-07:
I don't think it's a bug, you can't do that with precision drivers in general. I have a Lenovo Windows machine on which i never installed any touchpad drivers myself. The inbuilt trackpad has Windows Precision Drivers. When I click with two fingers and drag (which i never do but I tried it) it does exactly what you described, as if i was only pressing down with one finger. 
I think this is because the two finger click was a feature implemented to help macOS people get used to Windows (double click instead of right click was invented by Apple.) 

TL;DR Although Windows precision drivers have implemented the two finger click, I would advice you to just use the right click instead, because it is the "original" way to do it and I don't think it's a bug that the developper of this can solve (I don't think it's a bug at all but rather an incomplete feature.)

TorutheRedFox, 2022-05-30:
for me on a MacBookPro9,2, two finger dragging is not possible at all

the mouse just won't budge until you trigger a mouse up, opening the context menu, or whatever releasing a right click is programmed to do in a given window

Kleysley, 2022-06-02:
Press down with two fingers hard enough to engage the mechanical force touch built in to the trackpad. Drag your item(s) to the desired destination and then release the pressure off the button

TorutheRedFox, 2022-06-02:
there's no such thing as force touch on a mid 2012 macbook pro

Kleysley, 2022-06-02:
Im sorry I didntmean force touch meant press down with 2 fingers on the touchpad so it produces a physical click. Then let go one finger and drag it with the other. Does that work?

TorutheRedFox, 2022-06-02:
no that just triggers the button up event for the right mouse button, making the mouse act like I'm not pressing anything

closed 2022-12-30 by Etern213