imbushuo/mac-precision-touchpad #369: palm rejection (16" MBP)

kevinlong206, 2020-11-15:
**Pre-report checks**
- [X ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**

Greetings, after installing the driver I noticed what I initially perceived as erratic behavior. The most obvious example is while two-finger scrolling on a web page, the gesture would be interpreted as a 3 finger scroll which would cause the "show all windows" action in Windows. After a while I realized this is because my palm is wresting ever so slightly on the bottom of the 16" MBP trackpad, which extends almost to the front edge of the computer. 

This behavior does not occur in macOS, I believe they have implemented some palm rejection for the bottom X percent of the touchpad?

photo for reference: https://imgur.com/a/nEWVxYc

**Describe the solution you'd like**

rejection of bottom X percent of touchpad when other higher points are being used for gesture

**Describe alternatives you've considered**

Currently, I just had to told my hand in way that I am not accustomed to or use external mouse

**Additional context**
n/a

Thanks for consideration
- Kevin

logaandm, 2020-11-23:
I have the same problem. Palm rejection is non-existent on my MBP 2019 16". This makes the driver un-usable and I've uninstalled it.

sebastiaanvaz, 2020-11-23:
I also have the same issue. To me the issue mostly occurs when single-tapping, mistaking it for a double tap due to my palm resting slightly on the bottom of the touchpad. A fix would be awesome.

BillChen2K, 2021-03-21:
Same here. One way to reduce misoperation is to disable Tap with two fingers to right-click and Tap with a single finger to single-click in the Touchpad Settings panel.