imbushuo/mac-precision-touchpad #370: Mid-2009 MBP Kernel-Mode Drivers

ArturPortella, 2020-11-23:
Hello guys!

I'm currently working on User-Mode drivers to PID 0236 trackpad. The huge problem is that it's core 2 based CPU and all kind of stuff like browsing web or opening applications stutters or lag the cursor because of user mode driver. Can you guys port from UMD to KMD for this device? I know it's old and USB based, but it would be awesome to have less lag. The lag is so intense that I barely can work at the computer while doing another stuff.

Thank you in advance, and congratulations for this project! I would like a lot to share my time developing stuff like this, but I'm really focused at C++ development and the documentation for kernel drivers are very large...

imbushuo, 2020-11-25:
@ArturPortella porting UMDF to KMDF almost requires no code changes (except some WPP and USB params.) You can take a look at the Km driver for T2 and get some idea about it.

HenkPoley, 2021-03-11:
@ArturPortella Ah cool, setting WUDFhost.exe to Real-Time in Task Manager solves some of the stuttering 👍

imbushuo, 2021-05-30:
As an interim solution, I raised host driver priority: https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2105-3965

closed 2021-05-30 by imbushuo