imbushuo/mac-precision-touchpad #371: Porting current HID mini driver to filter driver?

imbushuo, 2020-11-25:
_Sigh_ Controlling over the whole Bluetooth BRB and L2CAP channel adds much more cost to the existing driver (and that's why Bluetooth release delayed for so long.) Try to explore the HID filter option, and see if it is possible to attach two upper filter drivers to the IO stack.

Rastafabisch, 2021-01-30:
I just came across [this project, which already features a Bluetooth hid driver](http://extramagic.forbootcamp.org/index.htm). While it’s free I do not know wether it’s open source.

duckyb, 2021-02-12:
> I do not know wether it’s open source.

It's not open source, unfortunately

imbushuo, 2021-05-29:
https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_wiki/wikis/Apple-PTP-Trackpad.wiki/47/Hijack-HIDCLASS

imbushuo, 2021-05-31:
Close this one, since PoC succeeded.

closed 2021-05-31 by imbushuo