imbushuo/mac-precision-touchpad #374: Could Spoofing the Brydge W-Touch in bluetooth settings grant us bluetooth for MT2?

raremotas, 2021-01-01:
Hello, I realize that the Brydge W-touch seems to be the only Precision touchpad with BT support. Would spoofing the device so windows uses BT for precision work? What kind of work needs to be done in terms of reverse engineering the BT signal that MT2 puts out? Is the data different than what you get through the lighting cable?

Thanks for your work on this amazing project. I am curious about the possibility of this product helping out development on bluetooth connectivity.

https://www.brydge.com/products/w-touch

mussemou, 2021-01-02:
Hey I was also thinking about this. I wonder if there is any workaround for Apple Magic Trackpad 2 also?

D0M1N4T0R99, 2021-02-15:
Isn't here any developer, who could achieve this? I am very looking forward for Bluetooth.
It's a great driver though.

closed 2021-06-01 by imbushuo