imbushuo/mac-precision-touchpad #375: Magic Trackpad 2 - registering touch when hovering very close to touchpad

adventures, 2021-01-04:
Hello, someone posted about this in July. (See https://github.com/imbushuo/mac-precision-touchpad/issues/319)

Is there any news about when this might be fixed? I have RSI and the magic touchpad is the only device that is tolerable on my PC. The unwanted trigger of the pad while hovering or even dropping finger to start movement causes the cursor to jump slightly and throw off movement. Thank you very much!

midjoin890, 2021-01-10:
Hi, I have the same problems and already tried solve it by decreasing sensitivity in Windows settings for touchpad by set sensitivity on min level, but it's not enough.

bugf1x, 2021-01-17:
same problem here. The sensitivity option has no effect i believe? "Input sensitivity configuration" is the last item on the roadmap.

azfarhashmi, 2021-02-16:
For me cursor jumps to bottom left like it do on macos sometime. Don't know exactly if it is similar or different issue then the title.

adventures, 2021-02-21:
> same problem here. The sensitivity option has no effect i believe? "Input sensitivity configuration" is the last item on the roadmap.

It seems to have no effect for me as well. Thanks for your input.

adventures, 2021-02-21:
> Hi, I have the same problems and already tried solve it by decreasing sensitivity in Windows settings for touchpad by set sensitivity on min level, but it's not enough.

It's the same thing for me as well.

hexagramg, 2021-02-21:
That's the same for me as well, unfortunate

adventures, 2021-03-13:
> That's the same for me as well, unfortunate

Sorry to hear that.

funky-jojo, 2021-03-17:
Putting my vote in here - excited for a fix.  I am having the same issues with the trackpad registering touches and movement when finger is hovering, but not touching.   Other than that issues, this driver is amazing, and works very well.

adventures, 2021-03-19:
> Putting my vote in here - excited for a fix. I am having the same issues with the trackpad registering touches and movement when finger is hovering, but not touching. Other than that issues, this driver is amazing, and works very well.

I completely agree, thanks for your input.

InIMoeK, 2021-03-23:
I'm having the same issue.

chenz97, 2021-03-25:
I'm having the same problem. Lowering the sensitivity doesn't seem to help.

taterujp, 2021-03-25:
#237 is also a duplicate of this issue.
I have stopped using Magic Trackpad 2 on my Windows 10 PC due to this issue and waiting for the fix.

> Putting my vote in here - excited for a fix. I am having the same issues with the trackpad registering touches and movement when finger is hovering, but not touching. Other than that issues, this driver is amazing, and works very well.
I also completely agree with this.

adventures, 2021-04-05:
> #237 is also a duplicate of this issue.
> I have stopped using Magic Trackpad 2 on my Windows 10 PC due to this issue and waiting for the fix.
>
>
> I also completely agree with this.

Thanks for the link!

nickgarver, 2021-04-13:
yea love these drivers so much<3 thank u. having the same issue though. also freezes after typing for like 1second(or until i click the touchpad) thank you again though. amazing work and looking forward to an update.

msavazzi, 2021-04-13:
Same for me but note that the same issue is on the Mac... I tend to think is apple's fault.

For the rest: amazing driver, you saved me :) I use only trackpads and there are just a few now and this is the best I've found to use with W10... but without driver was useless...

gdsandoval, 2021-05-25:
I would love to see this issue get fixed. I am currently using the magic trackpad 2 myself after trying a few different trackpads. This is seems to work the best but this sensitivity thing can get frustrating. Is there a timeline for this? I know the devs have lives other than making a driver work for a non pc device but just wondering.

taterujp, 2021-06-02:
I am testing Release 3979 released just today with Bluetooth connection but confirming that this issue has not been fixed yet in this release. (Bluetooth connection is very nice and working perfectly!)

By the way I guess the reason of why not everyone is reporting this issue is that it happens only with prople whose fingers' capacitance is a bit higher than usual. :)

gdsandoval, 2021-06-03:
I haven't tried the new release yet. But I have been using the trackpad with a macbook pro because the sensitivity issue messes me up sometimes and moves files or clicks something I don't intend to click. I do not experience this while using it with a macbook. Thanks for all the hard work, I really do appreciate being able to use this on my windows machine.

adventures, 2021-06-09:
> I am testing the release 3979 released just today with Bluetooth connection but confirming that this issue has not been fixed yet in this release. (Bluetooth connection is very nice and working perfectly!)
> 
> By the way I guess the reason of why not everyone is reporting this issue is that it happens only with prople whose fingers' capacitance is a bit higher than usual. :)

Thanks for this!

imbushuo, 2021-06-12:
I started to work on a final solution for that, which eventually will leave the threshold adjustable. Some work will be done in `acceleration` branch before I sign off these features into main branch.

adventures, 2021-06-16:
> I started to work on a final solution for that, which eventually will leave the threshold adjustable. Some work will be done in `acceleration` branch before I sign off these features into main branch.

This is amazing! Thank you very much!

jjoshrtca, 2021-06-16:
> Putting my vote in here - excited for a fix. I am having the same issues with the trackpad registering touches and movement when finger is hovering, but not touching. Other than that issues, this driver is amazing, and works very well.

***I THINK I SOLVED IT!  My theory is in the last paragraph of my post***

Same here, @funky-jojo :( I think this may be related to another very specific issue (which can be easily replicated by right/two-finger clicking and opening several new browser tabs from a link tree (or even, say, references at the bottom of a wiki if they're lined up just right)).

I have to open several links in new tabs in a very tight, close-together vertical list for my job.  Even when I'm *very* deliberate and *completely* remove my middle finger from the trackpad (the main one I scroll with as it's the most natural position leading to the least repetitive stress pain), the mouse cursor will sometimes very slightly jump and I will accidentally left/one-finger click the link because I'm trying to get all these links open in new tabs so fast (I'm such a good, little worker :) ).

Somebody mentioned some people's hands having more capacitance than other's.  I do love a the hand lotion, so my hands are *always* soft and supple and, hence, moisturized :P but I just completely washed them with regular softsoap (I think that's the brand name) and rinsed them real good with cool water.  I then dried them off with those unbleached paper towels and used the damp paper towel to completely wipe down the trackpad.  Then, I used a soft, cotton cloth to clean the trackpad with a little more force to be sure I got everything off of it (even though I just bought it last night).  Lastly, I went one step further and rubbed some 50/50 rubbing alcohol and water onto my hands and let them completely dry for a couple minutes.  They are now dry af and I *still* have the same problem (both hovering and jumping).

***I was looking at how close my fingers had to be to cause the hovering problem and (thanks to some good natural lighting) noticed that the body heat from my fingers causes a very slight, short-lived sort of "fog" on the trackpad that slowly dissipates over the course of a second or two.  What I noticed was that the hovering/jumping issue did not appear at first, but after my fingers got slightly down the trackpad a second or two later (I was trying to replicate another issue caused by the hovering that can be best described as a sort of phantom scroll).  What I noticed was that the phantom scroll caused by the hovering coincided with the fog on the glass of the trackpad.  I think this is the root cause of the problem.***

(I hope this helps :) Thank you to the dev/s for all your hard work in making this awesome driver!  Feel free to let me know if you'd like a screencap of what I'm talking about in my first paragraph.  I'd be more than glad to send.)

gdsandoval, 2021-06-16:
Same with me. I even used a screen protector from a cell phone but it was
the same. It's like the touchpad sensitivity is maxed out. I also noticed
the speed is very fast as well compared to when being used on os x.

andymcarter, 2021-07-12:
With the latest Bluetooth release I have not noticed this problem occuring, previously with wired I had this occasionally. Generally the control is a LOT more stable on BT that wired :) almost like having my old mac back but on Windows.

sonoks01, 2021-09-04:
Hi, I am having the same issue.
This is the best driver but this issue makes the touch Pad very difficult to use.
Please, I am waiting for the updates/fixes.

richardoSGSI, 2022-05-20:
new user, still experiencing the phantom mouse inputs as others described whether by direct connection via usb or bluetooth