imbushuo/mac-precision-touchpad #378: Installation has no effect, 2015 MacBook Pro, 13-inch w/ retina display

hmstve, 2021-01-13:
**Pre-report checks**

- [check] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
When I install the latest version, and go into precision touchpad settings, the only option there is for touchpad sensitivity, and nothing else. 

**To Reproduce**
Download the file, extract it, install the AmtPtpDevice file, and go check settings. 

**Expected behavior**
I expected there to be many more options inside the settings. However, it is all but blank, and none of the default trackpad gestures work either. 

**Environment**
MacBook Pro from early 2015, 13-inch with retina display.

hmstve, 2021-01-13:
Also, I know that the 2015 MacBook pro is on the list of devices that have errors, so its probably that, but I hope you can help me out.

imbushuo, 2021-01-20:
May I have your device ID?

hmstve, 2021-01-21:
> May I have your device ID (USB VID/PID, SPI device path)?

Hey,
I'm sorry, but in recent days my trackpad has become completely
nonfunctional, and I have decided to simply get a mouse. Still, here is
what popped up. SPI\VID_05ac&PID_0273&MI_02. As I am not very well-versed
in tech, I am not sure if I have gotten the right one, so I have attached
an image. Please tell me which one to do if this one is incorrect. Thanks
for making this thing by the way, I really appreciate it.
Thanks,
hmstve

imbushuo, 2021-01-25:
This one is supposed to work. It might get overlapped with the USB config. Does it work with previous version?

---
Thanks,
Bingxing Wang

hmstve, 2021-01-27:
Before my trackpad broke, I did try several previous versions, but none of
them worked either. If it is confirmed to work with a computer like mine,
it is probably just something I didn't do right on my end.

Thanks,
hmstve

esetnik, 2021-02-24:
@imbushuo  I have the same issue on 2018 Retina Macbook Pro. Installation doesn't do anything on the latest release version.

esetnik, 2021-02-27:
> @imbushuo I have the same issue on 2018 Retina Macbook Pro. Installation doesn't do anything on the latest release version.

I figured it out. I accidentally used the arm not amd release package.

pgumiela, 2021-03-06:
@imbushuo, maybe let's close this issue if it cannot be reproduced or just wrong version was installed?

closed 2021-03-07 by imbushuo