imbushuo/mac-precision-touchpad #383: navigate forward/back

dongyj1987, 2021-03-22:
can't navigate forward/back after setup in "advanced gesture setup"
"高级手势设置“中将三指或四指的左滑/右滑手势设置为”向前导航“”向后导航“后没反应

chenz97, 2021-03-29:
Works for me with a Magic Trackpad 2 on Microsoft Edge. Custom shortcut recording didn't work for me until a reboot. Maybe you should try rebooting the device and setting again?

dongyj1987, 2021-03-29:
Maybe I didn't express my situation clearly. It also works for me on Edge or Chrome with the same device. The custom shortcut gestures also work except for the "navigate forward/back" gestures. I think these 2 gestures should work in windows file explorer but they don't. I've tried a reboot but the problem remains.

抄送: " ***@***.******@***.***&gt;;

chenz97, 2021-03-30:
Tried in file explorer and it works for me. Maybe that's a bug then.

dongyj1987, 2021-03-30:
I've tried MagicUtilities and ExtraMagic before this small program, maybe it has something to do with them but I don't know how to clear them completely .

wcsherrod, 2021-03-30:
> 
> 
> Works for me with a Magic Trackpad 2 on Microsoft Edge. Custom shortcut recording didn't work for me until a reboot. Maybe you should try rebooting the device and setting again?

How are you able to record custom shortcuts?

I am not able to go forward or back (at least not in the standard way by two-finger swiping). I also had a previous installation of ExtraMagic, but I uninstalled it and removed the Apple driver before installing. Other gestures seem to work as expected, although I would prefer to remap some to custom key combinations like in macOS with BetterTouchTool.

dongyj1987, 2021-03-31:
There is a "advanced gesture setting"(translated from Chinese by myself) button on the right in the touchpad setting page where you can record your custom shortcuts.
I think we met the same bug that is navigating forward/back does not function except for in the browser. Maybe it has something to do with the previous installation of Extramagic or MagicUtilities but I'm not sure. I remaped the shortcuts of Alt+left/right in keyboard to the swiping left/right gestures with 2/3/4 fingers to solve it temporarily.

wcsherrod, 2021-03-31:
Thanks, that is very useful. Unfortunately, it still doesn't solve the two finger swipe navigation problem. Under advanced gesture configuration, there are only options to configure three and four finger gestures. I don't see anything in the touchpad settings that refers to two finger gestures for forward and back.

dongyj1987, 2021-03-31:
You can use MagicUtilities to reach your target. But I don't know why the scrolling speed is too fast so I gave up.

wcsherrod, 2021-03-31:
I just tested this in Edge and Chrome, and it seems to work. It does not work in Firefox or file explorer though.

Mapping three finger swipe to forward and back navigation under advanced gestures does work in FF and everywhere else. The problem seems to be recognizing two finger gestures specifically.

wcsherrod, 2021-03-31:
Apparently two finger swipe to navigate back works in file explorer for at least one other person, so I don't see why it shouldn't work there for us.

I have tried Magic Utilities, but I agree that the scrolling is awful and customization is pretty limited. I'm not a fan of renting basic software utilities like this.

wcsherrod, 2021-03-31:
Update: I just tried a fresh install on another machine that has never been connected to the trackpad. The behavior is the same. swiping with two fingers to go back works in Edge and Chrome, but not in Firefox or file explorer. Everything else works. The first machine is running Win10 Pro 20H2 19042.867 and the second is Win10 Home 2004 19041.867.

dongyj1987, 2021-04-01:
My solution is remapping the alt+left/right to the swiping left/right gestures with 3 fingers. Anyway it works and you don't have to move the cursor.

wcsherrod, 2021-04-01:
I’ve been doing the same thing for now. I would be interested to hear more from anyone who is able to get two finger swiping to work across all programs though.

imbushuo, 2021-06-01:
It doesn't work with Firefox because Firefox doesn't implement the required API.

closed 2021-06-01 by imbushuo