imbushuo/mac-precision-touchpad #388: Single touch (click) is way too sensitive - should have to press hard enough to hear the click

ptbtx, 2021-05-10:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**.
Single touch (click) is way too sensitive - should have to press hard enough to hear the click.  I have the sensitivity set to least sensitive.  The other sensitivity settings don't seem to change the sensitivity.

**To Reproduce**
touch the pad just a little, not enough to hear the click, click will happen unexpectedly

**Expected behavior**
touch the pad enough to hear the click, click will happen expectedly

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Macbook Air (13 inch 2017) Big Sur 11.3.1

ptbtx, 2021-05-10:
p.s. Running latest Win 10 20H2 in Bootcamp.

remingtonchan, 2021-05-31:
Hi @ptbtx,

I think what you want is to disable the tap functionality of your touchpad. I just realized this now.  In my MacBook, I wasn't really a fan of tap to click, so that's to be disabled.

[image]
What you want to do is to actually disable these, so that you are required to "physically" click on your trackpad to register clicks!

Please let me know if it works for you!

Thank you!

ptbtx, 2021-05-31:
That fixed it. Thanks!

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

HolgerGerhardt, 2021-07-06:
> **Describe the bug.**
> … The other sensitivity settings don't seem to change the sensitivity.

I can confirm that changing the “tap to click” sensitivity settings in Windows 10 (“Most sensitive”, “High sensitivity”, “Medium sensitivity”, “Low sensitivity”) does not have any effect. I tested this on a Lenovo ThinkPad T490s with Release 3979 of this driver.

(Apart from that, everything works great, over both Bluetooth and USB. Thanks a lot for providing this driver!)

thomasfiala, 2022-02-12:
For me, changing the above mentioned setting does not change anything (Windows 11 build 22000 on Dell XPS 15 9510).
Is there possibly a way to compile a driver version that disables "tap to click" completely?