imbushuo/mac-precision-touchpad #394: can not install the inf file on windows 10

evolfish, 2021-05-28:
I'm not sure why I can not right click the "install" on the inf file. I can only right click the "Open" and the file would be open as text file in Notepad.
I'm on 2019 macbook pro 16" and installed windows 10.
does anyone have the same issue?

Naozumi520, 2021-05-30:
Try to install the inf file in Device manager: 
Open the "Human Interface Devices" slots and right click the "Apple Multi-Touch" device, click update driver, 
"Browse my computer for driver", "Let me pick from a list of available drivers on my computer", 
click "Have Disk", "Browse" and pick your inf file. Press ok and click "Next", then you should have the driver installed.
[image]

closed 2021-05-30 by imbushuo

evolfish, 2021-05-30:
@Naozumi520 
Thanks for your solution!
However, I don't know why I still encountered problems...
[image]

Naozumi520, 2021-05-30:
Hmmm, how about put the inf file in other places?

Naozumi520, 2021-05-30:
This may cause by the file path name is too long.

evolfish, 2021-05-30:
I pull the file to the download file. This should not be the problem (~_~)

imbushuo, 2021-05-30:
Mind sending setupapi.dev.log to me? Should be relocated at Windows\Inf.

abulka, 2021-07-27:
I had the same problem - it turns out that I was browsing inside the `.zip` file via windows explorer's ability to navigate inside zip files - **but the zip file was not truly extracted to disk**. 

As soon as I properly unzipped the zip, I could right click on the extracted `.inf` file and `install` was now an option.