imbushuo/mac-precision-touchpad #398: Touchpad middle mouse click in VMware Workstation

sudo-sandw1ch, 2021-05-31:
**Pre-report checks**

- [ yes ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
Attempting to middle click (using the 3-finger click) doesn't work on VMware Workstation.  This does work in native Windows 10

**To Reproduce**
Create a Linux VM (I'm running CentOS Stream) within VMware Workstation.  Within the Windows 10 Touchpad Setting, under "Three-finger gestures" -> "Taps", select Middle mouse button.  Within the VM, highlight text in terminal, then proceed to use the Three-finger tap gesture to paste previously highlighted text.  It will not work, but works with a regular mouse middle click

**Expected behavior**
The Three-finger tap gesture should paste highlighted text into the window where the Three-finger gesture is utilized.

**Traces**
N/A

**Environment**
Windows 10 20H2
VMware Workstation with CentOS Stream
Magic Touchpad 2

imbushuo, 2021-06-01:
tbh I can't solve the issue with VMware workstation. I tested it for a while in VMware workstation, and seems like it's VMware's agent doesn't work well with us. Therefore I am closing the ticket right now, and I might open a support case with VMware.

closed 2021-06-01 by imbushuo

ylluminate, 2021-07-11:
@sudo-sandw1ch please let us know if you learn anything else and @imbushuo please definitely let us know if you do open a ticket. This would be really useful.