imbushuo/mac-precision-touchpad #402: Release 3979 - Bluetooth driver, trackpad 2 intermittently stops responding

jasonzerr, 2021-06-09:
Trackpad intermittently stops responding.  Two events reported in event log that could be correlated:

Bluetooth HID device  has reports that are bigger than supported.

- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="HidBth" /> 
  <EventID Qualifiers="32769">1</EventID> 
  <Version>0</Version> 
  <Level>3</Level> 
  <Task>0</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2021-06-09T20:47:57.4333127Z" /> 
  <EventRecordID>20650</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="4" ThreadID="29844" /> 
  <Channel>System</Channel> 
  <Computer>JASON-ASUS</Computer> 
  <Security /> 
  </System>
- <EventData>
  <Data>\Device\_HID00000006</Data> 
  <Data /> 
  <Binary>0000010002005000000000000100018000000000000000000000000000000000000000000000000028</Binary> 
  </EventData>
  </Event>

Bluetooth HID device  either went out of range or became unresponsive.

- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="HidBth" /> 
  <EventID Qualifiers="32769">2</EventID> 
  <Version>0</Version> 
  <Level>3</Level> 
  <Task>0</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2021-06-07T14:31:43.6299913Z" /> 
  <EventRecordID>20470</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="4" ThreadID="3148" /> 
  <Channel>System</Channel> 
  <Computer>JASON-ASUS</Computer> 
  <Security /> 
  </System>
- <EventData>
  <Data>\Device\_HID00000006</Data> 
  <Data /> 
  <Binary>00000000020050000000000002000180000000000000000000000000000000000000000000000000</Binary> 
  </EventData>
  </Event>

imbushuo, 2021-06-10:
I guess those are double-reports (the Linux driver already has a way to workaround them.) I will see what I can do here. For now, please power cycle the Bluetooth device as a workaround.

641bill, 2021-06-30:
Also stops responding, but still says connected in bluetooth settings, 
when multiple finger pulps(area larger than finger tips) are touching, 
using bluetooth connection, latest release. 
This won't happen when connected via cable.