imbushuo/mac-precision-touchpad #403: Apple stole your code

sirburpalot, 2021-06-11:
Apple has released a Boot Camp update that includes a precision touchpad driver. After having downloaded and tested it, I'm sad to report that it has the exact same bugs as your driver, which means that Apple most likely took your code without crediting you and may be in violation of GPLv2. Just thought I should let you know this.

imbushuo, 2021-06-12:
Hmm, it might be acceptable depending on the portion of the code because this project is dual-licensed: the user-mode USB version is licensed under GPLv2 because I took a look at the driver implementation in Linux kernel. In contrast, the later SPI and T2 versions are licensed under MIT license.) 

In fact, I've told some of Apple's people that they should properly (officially) support it 2-3 years ago. I would not be going to deal with legal stuff for this time being, and moving forward, this driver will still be maintained anyway since my primary use case these days is using Magic Trackpad 2 (either USB or Bluetooth) with an [extremely powerful Windows workstation](https://www.lenovo.com/us/en/thinkstation-p620) :)

closed 2021-06-12 by imbushuo

sirburpalot, 2021-06-12:
I don't know why Apple's website claims only T2-based Macs are supported: their precision touchpad driver works fine on my non-Apple PC with a Magic Trackpad 2. However, there's not much of a point in using it at the moment, since it works exactly like your open-source driver, as I said.

imbushuo, 2021-06-12:
@sirburpalot Interesting to know that. However, there are still a few cases exist, like SPI-based Trackpad (e.g. T1 users), so this will still be maintained.

sirburpalot, 2021-06-12:
Never said this driver should stop being maintained. Keep up the good work!

imbushuo, 2021-06-12:
@sirburpalot Just tried the Apple driver out. I wonder what the same bug you have encountered.

Regarding the mouse acceleration, I think it's a personal preference on the driver behavior and I will offer an option to cancel the acceleration. I personally prefer having it though.

sirburpalot, 2021-06-12:
@imbushuo

1. If you move the pointer with one finger and then, without lifting that finger, place an additional finger on the trackpad and perform a two-finger click, it is erroneously registered as a single-finger click.
2. Moving the pointer down by about 10px and then immediately clicking the trackpad without lifting the finger causes the pointer to jump back up.
3. Abnormally high sensitivity: I can move the pointer with my finger close to the trackpad but not quite touching it (which is not present on macOS or Linux).

imbushuo, 2021-06-12:
@sirburpalot I can guess why they have the issue - they are using the same metric basis to implement PTP on Windows. I have some ideas on workaround/resolve these issues, but my current work bandwidth cannot be 100% guaranteed.

martinsstuff, 2022-04-30:
From my experience using both driver versions, there are definitely some differences between the drivers which make me think it might actually be a new development.

1. The official apple driver does not have this bug: https://github.com/imbushuo/mac-precision-touchpad/issues/477
2. When using the software "Bluetooth Battery Monitor", you can trigger a battery level refresh at any point when using the open source driver, meanwhile the battery level is only reported once after connecting with the official driver.