imbushuo/mac-precision-touchpad #405: Feature 640: Sensitivity Adjustment Interface/Application

imbushuo, 2021-06-12:
Mirrored from https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/640

Need to implement a sensitivity app to address various user reports on sensitivity issues. This application should be able to adjust tip switch and confidence threshold, and optionally share the value to devs for reference and enhancement purposes.

This application should cover all supported models.

User feedback ticket #270; Bug report #221, #305, #268, #331, #93, #375, #237, #319, #86, #48, #288, #388, #276, #256

imbushuo, 2021-06-12:
Test Github board sync.

minhe7735, 2021-06-29:
[image]
windows 11 have this built in, might want to looking into that first

jjoshrtca, 2021-07-02:
> [image]
> windows 11 have this built in, might want to looking into that first

That's got more to do with tapping while typing than mouse movement sensitivity.  I know on my PC I had to turn this UP to the MOST sensitive setting because I need to be able to type and move/click the mouse rapidly in spreadsheets.  Windows 10 has a blurb above this setting's drop down box that says "turn DOWN this sensitivity if they (taps) activate while you're typing".

jjoshrtca, 2021-07-06:
Any ETA for the next update?  Is there a way I can donate to maybe help accelerate the process?

adventures, 2021-07-07:
> Any ETA for the next update? Is there a way I can donate to maybe help accelerate the process?

I second this!

gdsandoval, 2021-07-07:
> > Any ETA for the next update? Is there a way I can donate to maybe help accelerate the process?
> 
> I second this!

I would also like to donate for the time and effort it takes to get this trackpad working in Windows.

ylluminate, 2021-07-11:
Thank you - came to report this as it is affecting several of our users and excited to see this progress!

jjoshrtca, 2021-07-16:
Any update on the ability to donate?  I'd really like to donate if it would mean the release of the fixed/updated driver!

jjoshrtca, 2021-07-23:
@imbushuo just wondering, did you have to train the touchpad while developing the driver?  If that's the case, then could retraining with more fingers help (as in other people)?

imbushuo, 2021-07-23:
@jjoshrtca No trained has been done, it's just "I think it feels good" thing.

jjoshrtca, 2021-07-23:
> @jjoshrtca No trained has been done, it's just "I think it feels good" thing.

Oh, OK.  Well, if you need a beta tester for the next release feel free to lmk.  I do a lot of work/spreadsheet usage, which I think is likely a large chunk of the userbase (other than, say, creative/multimedia editing).  Those two bases are probably the majority of the users.  Not likely a lot of gamers gaming on trackpads.  
I screen record my work so I can go back and check on orders for my job if needed.  I could also record my finger movements on the trackpad using my iphone so you could see where the mouse jumps, dead-zones/stops responding, misinterprets two finger taps for one, etc.

ylluminate, 2021-07-29:
Thank you for volunteering @jjoshrtca. I have a few users on my end that complain about it not feeling good/right, so definitely giving some metrics like that, if @imbushuo can use them, will be wonderful.

jjoshrtca, 2021-07-29:
> Thank you for volunteering @jjoshrtca. I have a few users on my end (as I mentioned before) that complain about it not feeling good/right, so definitely giving some metrics like that, if @imbushuo can use them, will be wonderful.
> 
> This is an excellent tool, definitely don't want to be perceived as complaining or ungrateful. I think everyone wants to just have a "macOS-like" experience where possible, so a huge, huge thank you imbushuo in the first place for what you do and have done here!

Agreed!

jjoshrtca, 2021-08-27:
Is there a progress tracker for this bug fix?  Thanks!

sonoks01, 2021-09-04:
Hi, I have #375 issue as well.
This is the best driver but this issue makes the touch Pad very difficult to use.
Please, I am waiting for the updates/fixes.

jjoshrtca, 2021-09-09:
> Hi, I have #375 issue as well.
> This is the best driver but this issue makes the touch Pad very difficult to use.
> Please, I am waiting for the updates/fixes.

Hi @sonoks01 agreed!  It's been months.  You say this is the best, but do you know of any others?  Just wondering if there's anything I can switch to.  Thanks!

adventures, 2021-09-18:
Is there anything any of us non coders can do to help? I'm excited to have a fix!

sonoks01, 2021-09-20:
> Hi @sonoks01 agreed! It's been months. You say this is the best, but do you know of any others? Just wondering if there's anything I can switch to. Thanks!

There are a few other things you can try if you internet search it like 'apple trackpad windows', I like the 'magic trackpad' which cost some money after the trial period.

MagicUtilities, 2021-09-20:
@sonoks01 most likely means the "[Magic Utilities](https://magicutilities.net/magic-trackpad/features)".

jjoshrtca, 2021-09-20:
@MagicUtilities does it work on Windows?  I think there's another one called trackpad++ that is only for actual macs (which... why on earth create a third party driver when Apple already makes the best software/drivers anyways).  
If Magic Utilities is for Windows, then does it jump and jitter like this one?  I don't mind downloading trialware and paying if it works!
Thanks :)

MagicUtilities, 2021-09-21:
@jjoshrtca Windows, no jitter, simply try it, [our website is here](https://magicutilities.net/magic-trackpad/features) (but be aware it's for the Magic Trackpads only).

MagicUtilities, 2021-09-22:
@jjoshrtca Thank you. We just email you directly as any further [Magic Utilities](https://magicutilities.net/magic-trackpad/features) questions do not belong here.

gdsandoval, 2021-09-22:
The magic utilities app works, but it essentially turns the trackpad into a
regular mouse. It's not a real trackpad experience. Using this driver works
but that sensitivity issue creates a bit of frustration because sometimes
I'm moving something I didn't mean to move or scroll when I didn't want to.

jjoshrtca, 2021-09-22:
@gdsandoval it works like a regular touchpad.  @MagicUtilities app has dozens of settings to customize gestures and sensitivities and directions, etc. for all windows' gestures.

ylluminate, 2021-09-22:
Yes, MagicUtilities did not work out for us. We tried and tried, but it was too clunky in various ways. Mac Precision Touchpad is definitely the way to go, we just need more contributors to help out it seems. It's hard for open source labors of love like this to really resolve all of the issues and @imbushuo 	has done a heroic job of really pushing to where it's gotten at this point. Hopefully we'll see it track forward, but this is a rough one.

gdsandoval, 2021-09-22:
That is why I wanted to donate. To help get this going!

jjoshrtca, 2021-09-22:
@ylluminate just wondering, are you all trying to use this to game or draw or something?  Just wondering, because I'm using my magic trackpad 2 (2015) for work and in a program with cells that are literally like 32x16 pixels on a 20" FHD monitor, and I've had perfect performance out of @MagicUtilities app and with no jumps and jitters.  Hope this doesn't come across as adversarial or anything.  I'm genuinely interested.  Thanks :)

gdsandoval, 2021-09-22:
Not me. Just standard work. I am in the IT field and when I just use it as
normal, if any of my other fingers come close to the trackpad it registers
a touch. I can move the mouse just by hovering my finger above the trackpad
without even touching it.

jjoshrtca, 2021-09-22:
@gdsandoval that's what this driver -- mac precision touchpad -- does (jumps/jitters/hovers/etc).  @MagicUtilities app -- magic utilities -- does not do that.  That's why I was confused.  I was more wondering what @ylluminate found clunky about magic utilities.  Are there not enough gestures or something?

MagicUtilities, 2021-09-23:
Some technical insights of Windows driver development which hopefully helps to understand some aspects. 

When we started with development of the [Magic Utilities](https://magicutilities.net/magic-trackpad/features) version 3 years ago, Windows 7 had still a huge market share so there was no other choice than to make a real mouse driver otherwise we would not have reached most of our potential customers as PTP drivers need Windows 8 or higher.

Also a PTP doesn't do the actual touch processing it's done by Windows itself. The PTP only reports the touches, all mouse pointer movement, scrolling, etc. is done by Windows by processing the reported touches.

Mouse driver vs PTP driver has advantages and disadvantages on both sides:

### PTP driver advantages:

- You only need to report the touches properly (which is the main issue of this driver)
- You get any gesture detection for free
- You get the user interface for free

### PTP driver disadvantages:

- You are limited to the gestures Windows supports, i.e. no 3 finger drag, no 2 finger swipes, ...
- You can't control any aspects of any gestures, i.e. swap swipe directions, drag lock, non linear scroll speed/acceleration, individual gesture sensitivities...

### Mouse driver advantages:

- Windows 7 support
- Legacy mouse support for older software (look for other issues here)
- Full control of most aspects of all gestures
- Support for any gesture you can think about

### Mouse driver disadvantages:

- Shared global mouse properties with other mice, i.e. mouse cursor speed and scrolling speed.
- You need your own user interface
- All touch processing has to be done in the driver (this is the really hard part!)
- No native touch user experience for some gestures like changing the virtual Windows 10 desktop
- No smooth Pinch zoom gesture
- No momentum scrolling

The Magic Utilities will eventually be a hybrid PTP/Mouse driver. Certain touches will be reported to the PTP driver part, i.e. for momentum scrolling and pinch zoom, where other gestures will remain as a mouse driver. Ultimately the user can configure how he wants the gestures processed (as mouse or PTP).  But this is a long way to go as Windows kernel driver development is hard work! It must be bug free otherwise you get a BSOD.

For example we spent weeks alone to make the mouse pointer movement smooth and jitter free and even smoother than any other PTP implementation from builtin trackpads on Windows laptops. Only the native macOS pointer movement is still better.

As usual, there is no "One size fits all solution" and things are getting quite complex if you really look under the hood.

For now at least the user has a choice which driver he prefers.

jjoshrtca, 2021-09-23:
@MagicUtilities thanks for that writeup :) That's really interesting!

tritegeist, 2021-09-23:
For me the biggest downside to Magic Utilities is the lack of Momentum Scrolling. That and the subscription fee.

ylluminate, 2021-09-24:
@jjoshrtca I have had essentially the same scenario that @gdsandoval has had. As a matter of fact @MagicUtilities themselves recommended that I look at this particular driver as a solution to the problems that we had and that I was trying to address with them via tech support.  This driver has been superior in many ways and the near-native experience is nice, but indeed this issue is what needs to be addressed more than anything else. 

We use more than several computers with Magic TrackPad 2 devices that dual boot and if Apple somehow breaks OCLP at some point, these computers will become entirely Windows and/or Linux systems down the road (God bless the OCLP folks in their efforts to keep Apple from screwing those of us who have such hefty investments in them).  This leads us to want simplicity and openness in our drivers.  That is what this driver does and we'd simply like to see this driver not only survive, but thrive.

I'm sure imbushuo is super busy and it may be appropriate that others start contributing to help out.  At this point this is becoming a very sore spot, but I'm also not convinced MagicUtilities is the right solution either for reasons from the previous problems.

To be honest with you, I'm getting a sour taste with the marketing feel towards MagicUtilities that this thread has now achieved.  I have to say that I question incentives and sincerity and it's really turned me further away from trying MagicUtilities again... I think more harm than good has been done at this point.  ☹️

jjoshrtca, 2021-09-24:
I see where you're coming from.  That being said, I'm glad I found Magic Utilities and that they wrote up a thorough comparison and know both drivers' strengths and limitations.  It's also pretty cool that they sent you over here @ylluminate.  I see their write up was deleted, which is a bummer.  Maybe this driver's devs can point to Magic Utilities' driver on the project homepage for those who would benefit more from it, like business users who need precise mouse movement to click in very tine cells in spreadsheets and ERP software.  Humans don't always have to be in competition :) Most ERP software modules do not have an undo button and a single misclick can create 2x the work.  IIRC, I found this driver through a prominent tech site, so I'm sure this driver's got a good deal of users.
I'd still love to donate towards this project.  If money's not the devs' thing, then maybe they have a charity or organization that they'd be interested in directing users towards to donate.

tritegeist, 2021-09-25:
I've been debugging the driver all day today and I figured out how to solve the issue with the trackpad activating while your finger/s are hovering. This particular line of code seems to be causing the issue:

https://github.com/imbushuo/mac-precision-touchpad/blob/9d006db6f767806d46585b4241e81b0c89932ac3/src/AmtPtpDeviceUsbUm/InputInterrupt.c#L518

The problem is that using TouchMajor to determine the TipSwitch value doesn't work because TouchMajor values are registered even a finger is hovering above the trackpad. According to the Microsoft documentation [here](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/touchpad-windows-precision-touchpad-collection), tipswitch should be used to indicate whether or not the "contact" (i.e., finger) is touching the surface of the trackpad.

Instead, I think we should use the pressure value to determine the TipSwitch value: 

https://github.com/imbushuo/mac-precision-touchpad/blob/734b084b365b6168234049a8a61e569ae5ab19ca/src/AmtPtpHidFilter/include/HidDevice.h#L75

When you set the TipSwitch value like this, hovering close to the trackpad doesn't cause any sort of jittery or movement of the cursor:

`PtpReport.Contacts[i].TipSwitch = f_type5->Pressure != 0;`

I will put a pull request together.

tritegeist, 2021-09-25:
Alright here is the Pull Request:

https://github.com/imbushuo/mac-precision-touchpad/pull/436

@imbushuo could you take a look and approve it or give me some feedback? Thanks!

gdsandoval, 2021-09-26:
Oh man, this is awesome news. I can't wait to try it out.

gdsandoval, 2021-09-26:
Can I download the driver with this change so I can test it out and give any feedback?

tritegeist, 2021-09-26:
@gdsandoval I've attached a copy of the x64 driver I built from my pull request. If you need the ARM build instead let me know. I've only been able to test it on my Magic Trackpad 2, if you have a MacBook or an older Magic Trackpad it would be nice if you could confirm that my changes work for either of those devices as well. 

Because it's not officially signed you will have to bypass the signing requirement in order to install it. I've provided the steps below you'll have to follow to get it installed. Follow these steps at your own risk, disabling driver signature verification is dangerous and could allow very serious malware to be installed on your machine.

1. Disable Driver Signature Verification using the steps [here](https://www.howtogeek.com/167723/how-to-disable-driver-signature-verification-on-64-bit-windows-8.1-so-that-you-can-install-unsigned-drivers/)
2. Extract  mac-precision-touchpad-test.zip and right click on AmtPtpDevice.inf and click "Install" in the dropdown menu
3. Open Device Manager (click windows icon and search for "Device Manager"
4. In Device Manager, expand the "Human Interface Devices" section
5. Right click on "Apply USB Precision Touchpad Device" and click "Update Driver"
6. In the new window, click "Browse my computer for drivers" 
7. Click "Let me pick from a list of available drivers on my computer
8. In the list of available drivers, you should see one that says "Apple USB Precision Touchpad Device" and ends with "[9/26/2021]" - select that driver and click Next. Your touchpad should be using the new driver now. 

[mac-precision-touchpad-test.zip](https://github.com/imbushuo/mac-precision-touchpad/files/7232037/mac-precision-touchpad-test.zip)

gdsandoval, 2021-09-26:
Awesome! Thank you. I do have a macbook pro 2018. I'll try with that as
well.

oriolpetit, 2021-09-26:
@tritegeist are you saying the magic trackpad drivers also work with Magic trackpad 1? I thought this was not implemented as per https://github.com/imbushuo/mac-precision-touchpad/issues/358

tritegeist, 2021-09-26:
> @tritegeist are you saying the magic trackpad drivers also work with Magic trackpad 1? I thought this was not implemented as per #358

I was under the assumption that this driver worked for Magic Trackpad 1 as well. Magic Trackpad 1 is still unsupported, apologies for the confusion!

gdsandoval, 2021-09-28:
With the new driver the touchpad doesn't have the hover issue or
sensitivity thing going on. Will keep testing and using it to see how this
works out. I have only been using this on a Windows 10 x64bit system so far.

adventures, 2021-10-02:
> @gdsandoval I've attached a copy of the x64 driver I built from my pull request. If you need the ARM build instead let me know. I've only been able to test it on my Magic Trackpad 2, if you have a [...]

Wow... this is working beautifully! I've tested a trackpad 2 both wired and via Bluetooth, and currently have two trackpad 2s running smoothly via Bluetooth. Is there anything I can to to help test further?

Not to sound over the top, but I can't thank you all enough for all of this.. For those with chronic pain due to disabilities, this is truly special.

imbushuo, 2021-10-02:
Re: the PR, here's production signed build (note they are out-of-band build, not mainline release):

- [PTP-4187-AMD64-OOB.zip](https://github.com/imbushuo/mac-precision-touchpad/files/7271354/PTP-4187-AMD64-OOB.zip)
- [PTP-4187-ARM64-OOB.zip](https://github.com/imbushuo/mac-precision-touchpad/files/7271355/PTP-4187-ARM64-OOB.zip)

adventures, 2021-10-03:
> Re: the PR, here's production signed build (note they are out-of-band build, not mainline release):
>
> * [...]

Successfully installed two trackpad 2s via wired connection on a fresh install of Windows 11 beta. As I had two trackpads, I needed to manually update the driver for the second trackpad via tritegeist's steps 3-8 above.

gdsandoval, 2021-10-05:
If you have more than 1 finger on the touchpad, for instance if you were
dragging a window or something like that then the hover issue comes back.

egorgam, 2021-11-14:
I think it may be helpful point - if I zoom in web page in browser window (Edge for example), there are fuzzy micro twitchings while I dont't get off my fingers from Magic Traclpad 2. So it's two-finger case too

marc2323, 2022-01-27:
Commenting on @gdsandoval :

Having the same issue. PTP-4187 fixed the single finger hover jitteriness, but with two fingers it comes back.

jjoshrtca, 2022-08-25:
> Commenting on @gdsandoval :
> 
> Having the same issue. PTP-4187 fixed the single finger hover jitteriness, but with two fingers it comes back.

I recently decided to give this another try on a new pc and I still have single finger hover jitters.  Also still have 3/2 finger gesture fails as described here (https://github.com/imbushuo/mac-precision-touchpad/issues/488), and all kinds of other erratic behaviors like cursor jumps when trying to move in tiny spreadsheet cells or click close together links (ala an old fashioned site map link tree), cursor locks while trying to drag/move, etc.  There's a subscription based mouse emulator program for this trackpad that works perfectly, though.

provalinf, 2022-09-02:
Yes, jitter is fixed with 4187, but I find that touch is not sensitive enough now.
The surface must be rubbed at a certain threshold for the displacement to be taken into account.
I think it would take an in-between to make it perfect.

bradharms, 2022-09-04:
> 

Confirmed, it takes about 1mm of movement before cursor begins to move on a new Windows 11 install. (But haven't tested if this is something configurable with Windows settings.)

henusys, 2022-11-13:
@imbushuo Dear author, Can you build three or four versions with the lower, middle and higher sensitive settings, and ompressing it into a driver package. I think it would  solve the most people's sensitivity adjustment problems.
@ylluminate , @ylluminate

vitoplantamura, 2024-06-10:
Hi everyone,

I think I've created something similar to the spirit of this issue:

https://github.com/vitoplantamura/MagicTrackpad2ForWindows

This version has a Control Panel, haptic feedback control and additional options to increase pointer precision.

I didn't create a PR and I created a separate project because my project only targets the MT2 via USB (for the reasons explained in the main README of my project, in any case attributable to the well-known problems with official driver signing of the bluetooth version). Also, sadly, it seems clear that this excellent project by @imbushuo is dead - which is a shame as Apple's official 2021 driver for the MT2 is far from perfect.

Anyone who wants to test or comment on my project is welcome :-)

Thanks, Vito

pitonoved, 2024-06-10:
@vitoplantamura You're my hero! Great job!!!