imbushuo/mac-precision-touchpad #407: Release 3979 Bluetooth not working

michaelwangsz, 2021-06-13:
bluetooth not working

im using imac 16,2 windows 10 pro build 19042.1052
every time i reconnect bt (ie forget device then reconnect) it defaults to the apple multi touch pro driver
manually updating the driver by selecting your driver doesnt work and gives error 'This device cannot start. (Code 10)'
im sure none of them work ive tried every driver by bingxing wang

tldr bt driver doesnt work for me

usb works fine

also update the chocolatey thing cause it took me an hour to find out it is an outdated version, or just remove it and direct noobs like me, who follow step by step to download straight from github, to just download it from github and installing the driver

if this isnt helpful im a noob and only know basic stuff like how to update driver totally didnt just google and kinda but not really understanding ever line of code. thanks for your hard work and making this accessible to people like me unlike those who phrase their stuff such that only professionals know what they are talking about

imbushuo, 2021-06-13:
@michaelwangsz some migration thing needs to be done.

First, you need to specify the generic "Bluetooth HID Device" for root device BTHENUM{00001124-0000-1000-8000-00805f9b34fb}_VID&0001004c_PID&0265\7&139a387e&0&64A5C3F05EFA_C00000000, which is the "Apple Multi-touch Pro".

Then, install this driver for the "HID-compliant mouse" device. Make sure you have fully removed "Apple Multi-touch Pro".

This isn't a bug, but rather a current limitation that needs proper migration.

closed 2021-06-13 by imbushuo

michaelwangsz, 2021-06-14:
took me a while but now ive figured out.

0. Make sure you uninstalled Trackpad++ completely if you have previously does so
1. Go to the release tab in Github and download the newest version for your architecture (amd64 for intel macs/intel/amd/x86 **(i think)**, or arm64 for apple silicon M1 chip or later/snapdragon version of surface products) **(i think i really dk)**
2. Right click AmtPtpDevice.inf (or AmtPtpDevice without the .inf extension, its type is Setup Information) and click install
3. If you have a Magic Trackpad 2 and want to use it in Bluetooth mode: manaully pair it in PC Settings. make sure you can use the trackpad first (even if its bad)
4. go to device manager
5. click > beside human interface devices
6. right click the Apple multi touch pro, click update driver (or double click -> driver -> update driver)
7. click browse my computer for drivers -> let me pick from a list of available drivers on my computer
8. select "Bluetooth HID Device" then click next
9. give it like 5 seconds then Apple Multi-touch Trackpad HID Filter will appear when the device manager refreshes

@imbushuo put this up for people like me who dk half of what they are doing but wants to use magic trackpad2