imbushuo/mac-precision-touchpad #408: Can these drivers be used on Intel PCs?

getify, 2021-06-13:
Apologies if there's an obvious yes/no here I've missed. I have a surface laptop with intel cpu and windows 10. Unclear if/how to use these drivers in that scenario? Any guidance is appreciated.

imbushuo, 2021-06-13:
Yes. It is not restricted to Mac :)

[image]

closed 2021-06-13 by imbushuo

getify, 2021-06-13:
What I'm confused about is that the driver files listed here on github are named "arm" and "amd"... neither of which seem correct for an Intel system.

imbushuo, 2021-06-13:
@getify Intel EM64T is technically amd64, or called x64, so I am just using the build system convention there.