imbushuo/mac-precision-touchpad #410: Phantom scroll

jjoshrtca, 2021-06-24:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?  Yes

**Describe the bug**
A clear and concise description of what the bug is.
Hard to describe, but the two finger scroll gesture sticks around after scrolling is done and fingers are lifted off the tp.  This causes problems, e.g., in Chrome, this zooms out (or in, I guess, depending on how you have your natural scroll setup -- I have mine set to scroll down when my fingers move down for sake of consistency because it mirrors actual mouse behavior for the times when I need to use a mouse).

**To Reproduce**
Steps to reproduce the behavior:
Scroll, release, then immediately hit ctrl (like you're scrolling to the bottom to copy paste).  Easiest to see in Chrome where scroll+ctrl is zoom.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the scroll gesture to be gone when I lift my fingers off the tp.  I think this may be related to a lot of other issues like mouse jumping and hovering.  Maybe increased filtering/lowered sensitivity would help a lot.  

**Traces**
If applicable, add WPP traces to help explain your problem.
N/A

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.
Dell Inspiron 15 5000, USB w/ latest driver, Windows 10 (latest version/updates).

jjoshrtca, 2021-06-24:
Oh, forgot to mention that I think this may be due to the fog/condensation theory I mentioned in another issue about hovering/jumping.

imbushuo, 2021-06-26:
Redirect to feature #405

closed 2021-06-26 by imbushuo