imbushuo/mac-precision-touchpad #412: No longer working after Windows 10 Update

spectman, 2021-06-30:
**Describe the bug**
I can no longer get the Magic Touchpad 2 to work with my Surface Pro 3 after the Windows 10 build (19042.1052 os build). I can move the cursor with the trackpad but no gestures or multi finger clicks work, acts as if I do not have the driver installed.

**To Reproduce**
Steps to reproduce the behavior: 
Update to the newest version of windows 10 and then connect to trackpad via Bluetooth.

**Expected behavior**
Normal operation of this amazing driver, multi touch support, Bluetooth functionality.

**Environment**

Device name	DESKTOP-0SB0Q7Q
Processor	Intel(R) Core(TM) i5-1035G7 CPU @ 1.20GHz   1.50 GHz
Installed RAM	8.00 GB (7.60 GB usable)
Device ID	9C23C6F1-61FD-4DDD-8AB2-34DC5568DEA9
Product ID	00330-63086-82757-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 25 touch points

Edition	Windows 10 Pro
Version	20H2
Installed on	‎6/‎30/‎2021
OS build	19042.1052
Serial number	008530504057
Experience	Windows Feature Experience Pack 120.2212.2020.0

I am using the newest AMD driver type

imbushuo, 2021-07-08:
No repro here. I think you can attempt to reinstall the driver, it looks like the driver has been overridden by system.

closed 2021-07-08 by imbushuo

spectman, 2021-07-08:
Yep, reinstall fixed it. My bad