imbushuo/mac-precision-touchpad #413: Battery level status in Settings/Devices/Bluetooth

lloydo3000, 2021-07-02:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
I would like to know the battery level of the Magic Trackpad 2

**Describe the solution you'd like**
Similar to other devices, a battery icon in Settings, and a notification to charge when battery is low
[image]

**Describe alternatives you've considered**
[This 3rd-party software works](https://www.bluetoothgoodies.com) and reports the same battery information in Settings for my other devices. So I think the information is available.

**Additional context**
Thank you for the driver!! I am very happy so far with MT2 performance on USB & Bluetooth :)

imbushuo, 2021-07-08:
I am looking into that. Not sure if that needs GATT (which is part of BLE) on Windows though.

jialrs, 2021-11-16:
@imbushuo Thanks for your great driver!
It is really an important feature to view the battery status 😀. 
Looking forward to the new version ~

Yudezhi, 2022-01-06:
Subscribe this request.

edizionimigliori, 2022-06-06:
This would really improve functionality! Has any progress been made with the battery level feature?

benfishbus, 2023-02-14:
This is the one shortcoming I've found thus far. You can buy a separate app to do this (Bluetooth Battery Monitor - $5 for a perpetual license), but having it  built in would be ideal.

seanm-s, 2023-08-03:
This would be a great feature

bogdysk, 2024-02-25:
Are there any plans for adding this feature / continue development? I guess if you start Patreon many would support such an effort.